AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1108: Virus found

mvkirichenko, 2023-11-24:
Hi, if you have a question about AlDente, please use the Discussions section of Github: https://github.com/davidwernhart/AlDente/discussions
Today my antivirus detected a miner inside the application. How can it be?
[image]

davidwernhart, 2023-11-25:
Thank you very much for making us aware of this issue.

I have taken a close look at this case and you are right, VirusTotal also shows that 1 of 63 Antivirus services (DrWeb - which I assume is also the antivirus you are using?) is marking AlDente as malicious.
After some testing, it seems AlDente is is marked by DrWeb already since version 1.21 we released in March 2023. One thing that was changed in this version was that we started to use the library Blessed to improve the helper installation process. The library is open source and on github: https://github.com/trilemma-dev/Blessed.
After compiling the latest version of AlDente without this library for testing purposes, DrWeb doesn't mark it as a trojan bitcoin miner.
I strongly assume that this is a false positive by DrWeb, which seems to wrongly classify the code signature of the library. This is also supported by the fact that all other 62 virus scanners on VirusTotal did not report anything and there is no increased CPU activity by AlDente which would indicate such a miner.

In any case, we will definitely contact the developers of Blessed about this issue and see if we can sort this out. We unfortunately can not quickly remove the library from the production app without loosing the ability to install the helper, but we will do our best to sort this out quickly as possible.

Thank you for your understanding and kind regards from Vienna,

David
AppHouseKitchen

MatthiasKerbl, 2023-12-18:
Hi @mvkirichenko,

Good news, we have just released AlDente 1.24.1 which does not include the Blessed library anymore. Therefore, AlDente does not get reported by DrWeb or any other virus scanner on VirusTotal anymore.

Best,
Matthias

closed 2023-12-18 by MatthiasKerbl