AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1110: Charge Limit Resets to 100% After Restart/Shutdown

SidSidSid16, 2023-11-26:
**Describe the bug**
Whenever I shut down or restart my Mac, the charge limit changes from my set value of 49% to 100%.

**To Reproduce**
Steps to reproduce the behaviour:
1. Set the charge limit to a custom value such as 49%.
2. Click the 'Shut down...' or 'Restart...' button and proceed.
3. When booting back into macOS, the charge limit would have changed from set value to 100%.

**Expected behaviour**
The charge limit should be persistent despite restarts.

**Environment:**
 - MacBook model: MacBookPro14,3
 - macOS version: 13.6.1 (22G313)
 - AlDente Version: 1.24 (Free)
 - List of all enabled settings in AlDente: Launch at login, Intel mode

**Additional context**
I've been using Al Dente Free for quite a long time and this has never been an issue. The latest macOS version has been the root cause (_I think_). Updating Al Dente 1.24 worked temporarily but after a few restarts, the faulty behaviour has returned as the charge limit keeps resetting to 100%.

JasSingh685, 2023-11-27:
Same here. I saw the change log that this was listed in the bug fixes but unfortunately it's still happening in Intel mode.

MacBook Pro 13inch, 2020 Intel
MacOS Version: 14.1.1 (23B81)
AlDente Version: 1.24 (Free)

Didge3, 2023-12-05:
Facing the same issue as described above.
MacBookPro16,1 
MacOS Version: 12.6.8 (21G725)
AlDente Version: 1.24 (Free)

MatthiasKerbl, 2023-12-06:
Hi everyone,

We thought we fixed it this time but it seems like it is still happening to some. It should be fixed with the next update soon.

Best,
Matthias

SidSidSid16, 2023-12-19:
Works great, thanks!

JasSingh685, 2023-12-21:
Can confirm issue is now fixed with 1.24.1. Thank you!