AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1120: Disable Sleep until Charge Limit not working

mojoba, 2023-12-04:
**Describe the bug**
Macbook goes to sleep while plugged in even if current charge is way below charge limit.

**To Reproduce**
Close laptop lid when charge significantly below charge limit. Plug into wall. Macbook goes to sleep and charge remains the same when lid opened.

**Expected behavior**
Macbook should stay awake until charge limit is reached if "Disable Sleep until Charge Limit" is enabled.

**MacBook (please complete the following information):**
- M2 Air
- Sonoma 14.1.1
 - AlDente Pro
 - 1.24
 - List of all enabled/disabled settings in AlDente: All enabled

ESMP, 2023-12-05:
Thank God I'm not the only one... I thought I was going crazy. I think this started happening after 1.24 update. Because it worked as expected before. Not sure if it's only a thing with Silicon Macs, though.

TolleyLikesRice, 2023-12-06:
I've got the same since some time after updating to 1.24. If plugging in my macbook with the lid closed, charging is not started has it has done previously, however it does seem to charge if the lid is closed after charging is started.
[aldentepro-debug-2023-12-06-21:16:54.pasta.zip](https://github.com/AppHouseKitchen/AlDente-Charge-Limiter/files/13591730/aldentepro-debug-2023-12-06-21.16.54.pasta.zip)

MatthiasKerbl, 2023-12-07:
Hi @TolleyLikesRice,

Thank you for the debug file. We could locate the issue and it will be fixed again with the next update.

Best,
Matthias

ESMP, 2023-12-18:
I am currently using 1.24.1 and the problem seems to have been fixed. Thank you very much!

closed 2023-12-19 by MatthiasKerbl