AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1121: "Stop charging when sleeping" feature is not working.

jundawiz, 2023-12-04:
I am using Aldente Pro on a M1 pro.

I set my battery limit to 50%. Then, I charged it up to 50%.
When I put my macbook to sleep with BTT, it keeps charging.
I know this because when I wake my macbook up after a while, the battery goes over the limit i.e. 83%.

Please help.
I just bought the lifetime license yesterday.

andrewcrook, 2023-12-07:
is it the same as this  issue

#1120

andrewcrook, 2023-12-07:
> M1 Pro

Might also be useful to post the macOS version details and the exact model of your Mac as there are differences between Macs

MatthiasKerbl, 2023-12-18:
> I am using Aldente Pro on a M1 pro.
>
> I set my battery limit to 50%. Then, I charged it up to 50%. When I put my macbook to sleep with BTT, it keeps charging. I know this because when I wake my [...]

Hi @jundawiz,

Do you have the setting "Stop Charging when Sleeping" activated?

Best,
Matthias

MatthiasKerbl, 2024-01-23:
Hi @jundawiz,

It's been a while since we last heard from you, so I'm gonna go ahead and close this issue for now. If you need to come back to it later, just let me know. I'm here to help and can easily reopen it for you.

Best,
Matthias

closed 2024-01-23 by MatthiasKerbl

sanitybit, 2024-09-01:
@MatthiasKerbl  Sorry to comment on a closed thread, but I wonder if this is caused by initiating sleep with a third party app? The op mentions putting it to sleep with BTT (Better Touch Tool).

I _think_ I am running into the same issue when using Jettison to eject external disks before sleeping (`Eject and Sleep Now`) in the screenshot below. 

[image].

I just put it to sleep, and during the brief gap from then until I disconnected it from power, it charged from my 70% limit to 71%, and the Magsafe 3 charger showed a green LED while it was sleeping.

Here are my charging settings:

[image]