AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1123: After AlDente uninstallation, charging continues even when battery at 100%

Zoxjib, 2023-12-07:
After following the AlDente uninstallation FAQ page to remove the software fully, the M1 MacBook Air (2020) charges to 100%, and remains in 'charging' mode, never switching to 'plugged but not charging mode'
[image]

Before using AlDente when the battery reached 100%, charging stopped and the icon used to change to the below:
[image]

**To Reproduce**
Steps to reproduce the behavior:

1. Not sure... Uninstall AlDente

**Expected behavior**
Charging should stop at 100%, revert to how it worked before AlDente

**MacBook (please complete the following information):**
 - MacBook Air M1
 - macOS version Sonoma 14.2 Public Beta, also tested on Sonoma 14.1.2 beforehand
 - AlDente Pro
 - AlDente Version 1.24
 - List of all enabled/disabled settings in AlDente: Everything disabled before uninstall

binary-signal, 2024-01-21:
I experience the same problem, battery won't stop charging even when battery level is 100% charged

MacBook Pro 16 M1 
macOS version Sonoma 14.2.1
AlDente Pro
AlDente Version 1.24.1

MatthiasKerbl, 2024-01-26:
Hi @Zoxjib and @binary-signal,

I have just been to an Apple Store and checked on a bunch of MacBooks that never had AlDente installed before. Every single one of them was at 100% and had the bolt icon in the battery.
[image]

I am not sure if Apple changed it in one of the macOS updates but this means that it has nothing to do with AlDente.
Also, you don't need to worry about the battery continuing to charge at 100%. The battery management system takes care of that.

Best,
Matthias

closed 2024-01-26 by MatthiasKerbl