AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1124: Audio Interference with AlDente Pro Connected to External Audio Interface

samasante, 2023-12-12:
**Describe the bug**  
Experiencing loud feedback/ground loop noise from speakers when using AlDente Pro to limit charge on an M1 MacBook Pro connected to a Focusrite Scarlett 4i4 audio interface and two LG UltraFine 5K monitors. This issue occurs consistently when the charge limit in AlDente Pro is lowered or the Macbook reaches the limit threshold.

**To Reproduce**  
1. Connect M1 MacBook Pro to Focusrite Scarlett 4i4 audio interface and two LG UltraFine 5K monitors.
2. Open AlDente Pro and set a charge limit.
3. While the MacBook is charging, lower the charge limit in AlDente Pro.
4. Observe that as soon as the charge limit is lowered (and the bar turns blue), a loud feedback/ground loop noise is heard from the speakers.

**Expected behavior**  
The charge limiting function in AlDente Pro should operate without causing any audio interference or noise through the connected external audio interface and monitors.

**Screenshots**  
[image]

**MacBook (please complete the following information):**  
- MacBook model: M1 MacBook Pro (Sonoma) 
- macOS version: 14.1.1
- AlDente Pro or AlDente Free: AlDente Pro
- AlDente Version: 1.2.4

**Additional context**  
This audio interference issue arises only when the MacBook is connected to the external audio interface and dual monitors, and the charge limit is adjusted in AlDente Pro. This is a significant issue.

MatthiasKerbl, 2023-12-16:
Hi @samasante,

Thank you for the detailed bug report and we are sorry you are experiencing this issue. Unfortunately, I am not sure if can do anything against it since we are already using Apple's mechanisms to manipulate the charging behavior. However, if we find anything, we will give updates here.

Best,
Matthias

agingorange, 2025-04-14:
Commenting so I get notified and to remind myself. I get intermittent digital artifacts when using AlDente, when my MacBook Air M1 uses a Rodecaster Pro Duo.

MacBook model: M1 MacBook Air
macOS version: 15.4
AlDente Pro or AlDente Free: AlDente Free
AlDente Version: 1.32

samasante, 2025-04-14:
@agingorange This is what I used to fix my setup: https://www.amazon.co.uk/dp/B000KUD2G4

agingorange, 2025-04-14:
I wasn't getting hum, though. Like I said, it sounds like digital artifacts. I tried it with a DI box with ground lift and no change. Seems your issue is unrelated to mine, then.

agingorange, 2025-04-14:
Never mind, it was two new, but faulty USB cables!

closed 2025-04-15 by MatthiasKerbl

MatthiasKerbl, 2025-04-15:
Hi @samasante,

Thank you for sharing your solution, I will therefore close this issue.

Best,
Matthias