AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1128: Top Up: Can it ignore Sailing Mode to hit my limit %?

Sea-Worker9552, 2023-12-15:
**Is your feature request related to a problem? Please describe.**
When topping up, it seems the only way to top up is to top up to 100% or at least higher than what sailing mode would drop you to.

**Describe the solution you'd like**
It would be nice to be able to top up to any X% you want. So for instance, a lot of people use 80% charge limit. Let's say I have a 10% or 20% sailing mode active. My battery coudl be at 71% or 65% for instance and it wouldn't charge. I can't charge to 80% unless I turn off sailing mode, or alternatively I charge hit top up to 100% but manually disable charging at 80%. It would be nice to be able to charge right up to your charge limit

**Describe alternatives you've considered**
Yes I can top off and manually stop charging at the desired percentage, or turn off sailing mode while doing top off to ensure I hit my target percentage reliably.

MatthiasKerbl, 2023-12-16:
Hi @Sea-Worker9552,

In which scenario do you need to charge to the upper Sailing Mode limit?
The Sailing Mode range is meant to be set in a way that you have enough charge for daily usage and if you need more you can use the Top Up feature.

Best,
Matthias

Sea-Worker9552, 2023-12-17:
Hey there. So sometimes when I'm about to go on a trip (e.g. go to the airport, flight, etc.) which can sometimes be a long day, I'd like to charge my laptop a bit more. Yes I have topped up to 100% before, but sometimes I wonder if I can just top off to the charge limit manually, and so that is my use case.

My personal setting is 80% charge limit, 15% or 20% sailing mode. That is fine for day to day use at the office, but when I want to charge before a trip, there's no clean way to top off to 80% instead of 100% without some manual intervention.

MatthiasKerbl, 2023-12-18:
Hi @Sea-Worker9552,

Thank you for the clarification.
Again, in general, the Sailing Mode range is meant to be set in a way that you have enough charge for daily usage and if you need more you can use the Top Up feature. However, if enough users want a way to quickly charge up to the upper limit of the Sailing Mode, we might include that in one of the future updates.

Best,
Matthias

Sea-Worker9552, 2024-06-14:
Running into this a little bit more as my daily routines are changing. I agree with your previous statement @MatthiasKerbl that sailing mode is meant to make sure you have enough charge for daily usage. As travel is returning, on the days I travel, I really would like a way to bump up charge without having to go to 100% but also without having to tinker with my sailing mode. As my laptop is mostly docked on a daily basis, it can take weeks sometimes before my 80% drops to 60% for sailing mode. I'm sitting at 62% right now and I have a flight to catch tomorrow, and here I am wondering if I can top off to 80% easily. There's no easy way without disabling sailing mode basically, but I would have to re-enable it later.

Perhaps another way might be a 3 way toggle for top-off. Top off to limit, top off to 100%, and then off?

bluehunt81, 2024-07-18:
I agree with @Sea-Worker9552. There are various scenarios where one could use the "top-up to upper limit". 
Assuming an upper limit of 80% and sailing mode of 20%. If you move a lot around the house/garden during the day, you might get to your desk in the evening with 61%. It would be great to leave the laptop plugged in over night and just click a button to charge to 80%; otherwise you either leave it to charge to 100% overnight (defeats the purpose) or hope that you wake up a bit earlier the next morning, and either use something power intensive so that you get below the lower limit of the sailing mode and then the normal "charge to upper limit" will kick in or click "top-up" and then constantly check the battery level so that you do not overshoot (again charging to 100 would defeat the purpose). 
For my personal use case, this is the single most important feature that is lacking from this otherwise great software.

MatthiasKerbl, 2024-09-03:
Hi everyone,

We have just released AlDente 1.28.1 which now allows to charge to the upper sailing limit. You just need to double click on the charge limit bar in the Pop-Up window. To get AlDente 1.28.1 you just need to update within the app under Settings/General/Check for Updates.
Please report back if it works for you.

Best,
Matthias

closed 2024-09-03 by MatthiasKerbl

bluehunt81, 2024-09-06:
@MatthiasKerbl the new feature works great and is exactly what I needed, but it would be great if the "activation sequence" was a bit more accesible. It has happened a few times that I do not position the cursor on the exact charge limit or i move my finger a little on the trackpad just before clicking and instead of activating the feature, I just change the charge limit value. 
Otherwise it's great to have the feature implemented. 
Thank you so much.

Sea-Worker9552, 2024-09-06:
Sorry I'm a little bit unclear, what am I supposed to double click on here? Double clicking on the blue bar just resets the charge limiter percentage. Double clicking Top Up doesn't seem to help.

[image]

bluehunt81, 2024-09-06:
@Sea-Worker9552 you actually need to double click on the "slider" (the one that represents the maximum charge limit). If you misalign the clicks by a bit, you will indeed move the slider to the new position and set a new charging limit, which happened to me quite a few times and was a little annoying

Sea-Worker9552, 2024-09-06:
Whoah, ok I wouldn't have thought to click right on the bar itself. In that case I agree with you @bluehunt81 that a more accessible and discoverable interface would be appreciated.

MatthiasKerbl, 2024-09-09:
Hi @Sea-Worker9552 and @bluehunt81,

Thank you for your feedback. We will consider it in future development and will try to make it more accessible.

Best,
Matthias