AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1131: Different charge rules and limits for wall power vs dock power

mojoba, 2023-12-17:
**Is your feature request related to a problem? Please describe.**
When I am traveling and sporadically charging from wall power, I find it more useful to set the charge limit to 100% and disable "Stop charging when sleeping".
When on a desktop dock with power passthrough, I find it more useful to set the charge limit at 80% and enable "Stop charging when sleeping".

**Describe the solution you'd like**
Set different charge rules and limits for wall power vs dock power. Dock devices may be detected.

MatthiasKerbl, 2023-12-18:
Hi @mojoba,

Thank you for the improvement suggestion.
This might be a good addition to the Shortcuts feature. We just need to figure out how to distinguish between different charges.

Best,
Matthias

DamonHD, 2023-12-18:
I would like to have thresholds (and maximum charging power) linked to charge source, eg to be gentler on my off-grid system and stop it cutting out.

Rgds

Damon