AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1132: "Stop charging when sleeping" prevents charge to a sleeping macbook even if it is below the charge limit

mojoba, 2023-12-17:
**Describe the bug**
"Stop charging when sleeping" prevents charge to a sleeping macbook that is below the charge limit.

**To Reproduce**
Let charge fall below charge limit. Close macbook and let it go to sleep. Connect power. Macbook will not charge unless lid is opened.

**Expected behavior**
AlDente should wake macbook to check charge limit when power adapter is connected even if "Stop charging when sleeping" is enabled. This will allow the macbook to charge till the charge limit, after which AlDente can let the macbook go back to sleep.

MatthiasKerbl, 2023-12-18:
Hi @mojoba,

The setting "Stop charging when sleeping" does exactly what it says, it stops charging when the MacBook is asleep even if it gets plugged in while it is asleep.
If you want your MacBook to charge to the charge limit while the lid is closed or while it would otherwise be asleep, you need to activate the feature "Disable Sleep until Charge Limit".

Best,
Matthias

mojoba, 2023-12-18:
Hi Matthias,

The setting indeed does what it says, however if the setting is disabled
then, as the app states, the charge may no longer be limited by the app.

I’m not sure what the solution is since, as the documentation states, the
app cannot limit the charge while the MacBook is asleep anyway.

Is there a wake signal the app can produce to periodically check the charge?

MatthiasKerbl, 2023-12-18:
Hi @mojoba,

> Is there a wake signal the app can produce to periodically check the charge?

No, macOS does not allow third-party apps to wake up the MacBook.

Which behavior would you like to have?

Best,
Matthias