AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1133: MacBook Pro 16 intel 2019 does not sleep with closed lid

PedroSousaSilva, 2023-12-19:
Bug:
Since installing the new OS 14.1.2, I believe AlDente does not allow  my MacBook Pro 16' Intel to sleep when the lid is closed. Usually battery keeps discharging till it reaches 0%.

**To Reproduce**
I used the following command at the terminal: sudo pmset -g. At first, it pointed out as "Handoff" as the culprit of preventing system sleep.
First report>
System-wide power settings:
 SleepDisabled		0
Currently in use:
 lidwake              1
 lowpowermode         0
 standbydelayhigh     86400
 proximitywake        1
 standby              1
 standbydelaylow      0
 ttyskeepawake        1
 hibernatemode        3
 powernap             0
 gpuswitch            2
 hibernatefile        /var/vm/sleepimage
 highstandbythreshold 50
 displaysleep         2
 womp                 0
 networkoversleep     0
 sleep                1 (sleep prevented by sharingd)
 tcpkeepalive         1
 halfdim              1
 acwake               0
 disksleep            10
pedrosilva@Pedros-MBP ~ % sudo pmset -g assertions
2023-12-19 09:29:54 +0000 
Assertion status system-wide:
   BackgroundTask                 0
   ApplePushServiceTask           0
   UserIsActive                   1
   PreventUserIdleDisplaySleep    0
   PreventSystemSleep             0
   ExternalMedia                  0
   PreventUserIdleSystemSleep     1
   NetworkClientActive            0
Listed by owning process:
   pid 171(WindowServer): [0x000076c800099738] 00:00:00 UserIsActive named: "com.apple.iohideventsystem.queue.tickle serviceID:1000072ff service:AppleHIDKeyboardEventDriverV2 product:Apple Internal Keyboard / Trackpad eventType:3"  
	Timeout will fire in 120 secs Action=TimeoutActionRelease
   pid 575(sharingd): [0x000078a600019885] 00:02:06 PreventUserIdleSystemSleep named: "Handoff" .

Even though I have always used handoff and never had problems with it. I believe, the significant difference in my system was indeed the installation of the new AlDente app, with all the new pro features plus the new MacOS update.

Either way, I disabled "Handoff" and run the command again, with the following results:
pedrosilva@Pedros-MBP ~ % sudo pmset -g
Password:
System-wide power settings:
 SleepDisabled		1
Currently in use:
 lidwake              1
 lowpowermode         0
 standbydelayhigh     86400
 proximitywake        1
 standby              1
 standbydelaylow      0
 ttyskeepawake        1
 hibernatemode        3
 powernap             0
 gpuswitch            2
 hibernatefile        /var/vm/sleepimage
 highstandbythreshold 50
 displaysleep         2 (display sleep prevented by com.apphousekitchen.aldente-pro)
 womp                 0
 networkoversleep     0
 sleep                1 (sleep prevented by com.apphousekitchen.aldente-pro)
 tcpkeepalive         1
 halfdim              1
 acwake               0
 disksleep            10

The report pointed now the AlDente app as the culprit.
I believe these problems did not happen prior to the update of MacOS. I am also sending a debug file.
[aldentepro-debug-2023-12-19-10:20:22.pasta.zip](https://github.com/AppHouseKitchen/AlDente-Charge-Limiter/files/13714548/aldentepro-debug-2023-12-19-10.20.22.pasta.zip)

**Screenshots**
If applicable, add screenshots or screen recordings to help explain your problem.

**MacBook (please complete the following information):**
 - MacBook Pro 16' Intel
 - macOS version 14.1.2
 - AlDente Pro - every feature enabled except Intel mode

**Additional context**
Add any other context about the problem here.

MatthiasKerbl, 2023-12-19:
Hi @PedroSousaSilva,

Looking at your debug file, everything seems normal.
What are your exact steps when your MacBook discharges to 0%? Do you have your MacBook plugged in?
Also, could you generate and send a debug file while or shortly after your MacBook discharges to 0%?

Best,
Matthias

PedroSousaSilva, 2023-12-19:
Hi Mathias,

I just literally close the lid and leave everything opened. I leave the Macbook unplugged.
I will try it again tonight and will send you a debug file.

Kind regards,
Pedro Silva

PedroSousaSilva, 2023-12-26:
Dear Mathias,

Problem seems to be unrelated to your app but actually with “Handoff” system from Apple.

Sorry for the inconvenience I may have caused.

Kind regards,
Pedro Silva

closed 2023-12-28 by MatthiasKerbl