AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1134: Crash when clickign existing schedule

techyogi, 2023-12-19:
**Describe the bug**
AlDente Pro crashes when viewing existing schedule

**To Reproduce**
Steps to reproduce the behavior:
1. Set a schedule -  mine is daily top up at 6am
2. restart computer
3. open al dente
4. click schedule tab
5. Crashes

MatthiasKerbl, 2023-12-20:
Hi @techyogi,

I am currently not able to reproduce this issue on my MacBook.
Please answer the following questions:
**Screenshots**
If applicable, add screenshots or screen recordings to help explain your problem.

**MacBook (please complete the following information):**
 - MacBook model (https://support.apple.com/en-us/HT201608)
 - macOS version (https://support.apple.com/en-us/HT201260)
 - AlDente Pro or AlDente Free?
 - AlDente Version?
 - List of all enabled/disabled settings in AlDente:

Also, please generate and share a debug file and a crash report when the crash happens again.

Best,
Matthias

techyogi, 2024-01-02:
Currently on an m3 max running sonoma 14.2.1 and aldente pro, installed via setapp. 

I run this with the following settings other than default:

1. Dock icon off.
2. schedule set - single daily schedule to top up at 6am and run on next available as needed.

Thats it. I set this up, quit aldente, re-start aldente, and typically i see the issue the very next time i run the app and click schedule.

Log enclosed...

[aldente.txt](https://github.com/AppHouseKitchen/AlDente-Charge-Limiter/files/13806616/aldente.txt)

MatthiasKerbl, 2024-06-18:
Hi @techyogi,

Are you still experiencing this issue?

Best,
Matthias

techyogi, 2024-06-19:
Nope.  Works well now

*BENJAMIN SANDERS*
*Entrepreneur and Technologist*
*E  ***@***.*** ***@***.***>
*T* 917.817.9495

closed 2024-06-19 by MatthiasKerbl