AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1135: [bug]: install Helper

rrain7, 2023-12-20:
**Describe the bug**
After I install the latest version app, and I install Helper, but I click the icon got the message:
[image]

however, I have install the helper as before
[image]
 

I use Macbook Air M1, OS version:  12.6.6

MatthiasKerbl, 2023-12-20:
Hi @rrain7,

For some updates, it is necessary to reinstall the Helper. This should be completely normal.

Best,
Matthias

rrain7, 2023-12-22:
thanks for your reply, I have uninstall the app, and reinstall it, then install the  helper tool, but the app still notice me install the helper

step1:
[image]

step2:
[image]

but I get the info as follow:
[image]

rrain7, 2023-12-22:
I reinstall the latest version app
[image]

MatthiasKerbl, 2023-12-28:
Hi @rrain7,

Thank you for the documentation. I am not sure if I understand it correctly.
Are you experiencing a loop of the helper installation even with version 1.24.1?

Best,
Matthias

rrain7, 2024-01-01:
> Hi @rrain7,
> 
> Thank you for the documentation. I am not sure if I understand it correctly. Are you experiencing a loop of the helper installation even with version 1.24.1?
> 
> Best, Matthias

Yes, it always tips me install the helper application even I click the [Reinstall].

Happy new Year :-)

MatthiasKerbl, 2024-06-18:
Hi @rrain7,

Any updates on this matter?

Best,
Matthias

rrain7, 2024-07-09:
> Hi @rrain7,
> 
> Any updates on this matter?
> 
> Best, Matthias

Hello, after I updated the app, there is no such issue. Thanks

closed 2024-07-15 by MatthiasKerbl