AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1136: Use Run the Calibration Mode to cause shutdown

yxd-python, 2023-12-22:
**Describe the bug**
Use Run the Calibration Mode，When the power is less than 15%, it will turn off immediately when the computer starts charging again.
My computer is a 2018 MacBookPro 15-inch，The model is A1990. macOS 13.6

MatthiasKerbl, 2023-12-28:
Hi @yxd-python,

I am not sure I understand your issue. Do you mean that when your MacBook has less than 15% of the charge left in the battery and you plug it in, it will immediately turn off?

Best,
Matthias

yxd-python, 2023-12-29:
No, I plugged in the power all the time and turned on the calibration mode. Under normal circumstances, when the discharge reaches 15%, it will restart charging, but at this time, the computer will suddenly shut down and restart.I just updated the latest AIDente Pro 1.24.1. I'll see if a similar situation will happen, and I will give you feedback later. Thank you.

yxd-python, 2024-01-18:
Sorry, after my test, there is something wrong with the battery of my computer. Thank you.

closed 2024-01-18 by yxd-python