AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1139: Constant hang after 1.24.1 update and scheduled-calibration discharge to 15%

wills11, 2023-12-27:
After SetApp updated Al Dente Pro to 1.24.1 and the app discharged the battery to 15% as part of a scheduled calibration, the app hangs. Force-quitting, restarting, etc. — none help. The battery seems to recharge to 16%, but never gets past that.

Most of the time, hovering over the menubar icon results in a spinning beach ball. Sometimes, the menubar icon responds, and I can even open the Settings panel—at which point it hangs completely. If I close the Settings panel, I can eventually reopen it—but again, no response.

Running MacOS 14.2.1 on a 2023 16" MacBook Pro with the M3 Max chip—and again, this was not the result of any OS update, etc. It seems to have started only when the discharge to 15% was complete, which was almost 24 hours ago.

The problem is that, as I have the SetApp version, I can't even access the release notes to see the support contact info. So I'm stuck with a laptop that won't charge above 16%—and even force-quitting AlDente doesn't allow it to charge!

Please help—wondering if there's any way to roll back to the last version for those who have the app via SetApp? Thanks in advance!

MatthiasKerbl, 2023-12-28:
Hi @wills11,

Thank you for the bug report. Unfortunately, I am currently unable to reproduce this issue.
Do you find any AlDente crash reports in the macOS Console app under Crash Reports?
Also, please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on [How to generate and share a debug file](https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/) on our blog.

> The problem is that, as I have the SetApp version, I can't even access the release notes to see the support contact info.

You can find the release notes with the support email address in the Setapp app under AlDente by clicking on the current version.

> I'm stuck with a laptop that won't charge above 16%—and even force-quitting AlDente doesn't allow it to charge!

To get out of this state, you have to perform an SMC reset. For that, you just have to shut down the MacBook and close the lid for 30 seconds. Afterward, when you turn on the MacBook again, it should charge normally again.

Best,
Matthias

wills11, 2023-12-29:
Thanks a million, Matthias—this is all extremely helpful!

I was able to resolve the issue shortly after posting by force-quitting, then restarting the app, and immediately stopping the automated Calibration cycle: to do so I clicked "Top Up", but that didn't work until I deleted the scheduled Calibration and turned off "Start Tasks at next Opportunity", then force-quit and restarted the program. Apologies for not updating my post earlier—it totally slipped my mind.

I'm pretty sure the issue was caused by the automated Calibration cycle, since I've never had any other issue with the app. And the Calibration should have taken no more than a few hours while using the computer, but it had been running for almost an entire day before I noticed the battery was at 15% and saw the menubar icon beachballing.

I did not find any Crash Report with Console, but I did figure out that I should probably save a debug file and did so shortly after. I will send it to you at the support email, which I located—no idea why I assumed it was accessed in the app. I will note the SMC reset for any future emergency, and of course let you know if it happens again. I had been doing biweekly Calibrations, and will probably set one up again, especially now that I have some confidence that I can get out the same problem should it arise again.

I so appreciate your prompt and helpful response—AlDente Pro is a fantastic app!

MatthiasKerbl, 2024-01-23:
Hi @wills11,

Thanks for the update and we are glad that you like AlDente. :)
Also, I will close this issue here since we are taking care of this issue through our support.

Best,
Matthias

wills11, 2024-01-23:
Thanks for the fantastic app and superb customer support!

wills11, 2024-01-25:
Matthias,

No sooner had you closed the ticket and I’d thanked you, than it happened again!

Same deal, I think. 

I initiated a Calibration (this time unscheduled). 
During the Calibration last night, I had to restart for a totally unrelated reason (to rebuild Launch Services)—this was after AlDente had performed the initial full charge to 100% and was midway through the discharge to 15%. After restart, the battery continued to discharge to 15%—but never began charging.
This morning, it was stuck at 15% (after having completed the initial charge to 100% and discharge to 15%), and clicking on the menubar icon for AlDente Pro caused it to beachball (with no other apps affected). 
This time I followed your instructions and shut it down and closed the screen for 30 seconds (since I’ve been running it in clamshell, I first opened the screen, then shut down and closed it). 
After resetting the SMC, I was able to open AlDente as usual—but “Limit” was stuck at 15% I was able to reset the upper limit (which was stuck at 15%

I’m attaching the debug log again—obviously I saved it after the SMC reset, as AlDente wouldn’t respond before.

I have no idea if any of these factors could be involved, but thought I’d at least mention them: 

I was running 100% in clamshell mode: when initiating the Calibration, during the restart, and while it was stuck at 15% and AlDente beachballed.
As mentioned above, I had to restart my MBP during Calibration—at the time,
I’m not charging with the MagSafe charger, as the MBP is powered through the 90W port on my Dell U3223QE monitor <https://www.dell.com/en-us/shop/dell-ultrasharp-32-4k-usb-c-hub-monitor-u3223qe/apd/210-bdph/monitors-monitor-accessories> using an Anker 515 USB4 cable <https://www.anker.com/products/a8487?variant=42177509589142> (rated for 240W). The cable is simultaneously carrying the video signal to the monitor, but in regular mode, meaning not in MST (Multi-Stream Transport) mode. 

Am I the only person who’s experienced this during Calibration? It definitely first happened the first time I ran a Calibration after the most recent update—but since I only run a Calibration every ~10 days, it’s hard to know if there was some other change to the system that contributed. If you think there’s any chance that the restart during Calibration or the power cable is involved, I can experiment in the future.

Sorry to be the bearer of buggy news—and let me know if you’d prefer me to post this on Github or to start a new ticket.

Warm thanks,

William

￼

MatthiasKerbl, 2024-01-26:
Hi @wills11,

Thanks again for the detailed bug report and I am sorry to hear that you experienced this bug again.
At this point, I am relatively sure this is caused by a bug that is triggered during discharge in Clamshell mode. We have already fixed it and will release a new update soon.
When the update is out, please let us know if it is fixed for you as well.

Best,
Matthias

MatthiasKerbl, 2024-02-09:
Hi @wills11,

We have just released AlDente 1.25 which includes a fix for this issue. To get AlDente 1.25 you just need to update within the app under Settings/General/Check for Updates.

Please report back if it works for you.

Best,
Matthias

closed 2024-02-09 by MatthiasKerbl

wills11, 2024-02-10:
Thanks—I will keep you updated!