AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1140: Please use my configured hardware battery percentage or system battery percentage for the charging limit percentage.

jerrysearch, 2024-01-02:
**Describe the bug**
When the current hardware battery level is above 80%, charging is restricted to 80%, and the hardware battery percentage option is enabled. Discharge is displayed continuously until it drops below 80%. After continuous discharge, disable the hardware battery option, and the displayed battery percentage remains at 80%.

**MacBook (please complete the following information):**
 - Apple M1 Pro
 - macOS 14.2.1
 - AlDente Pro
 - AlDente 1.24.1

MatthiasKerbl, 2024-01-23:
Hi @jerrysearch,

I am not sure what exactly your issue is. Could you explain it in more detail?

Best,
Matthias

jerrysearch, 2024-01-24:
Hi @MatthiasKerbl,

I have enabled 'Hardware Battery Percentage' in the settings and set the charging limit to 80%. The observed behavior is that the laptop continues to use battery power until the battery level displayed is approximately 74% (software battery percentage is 80%). Only then does it switch to using external power. It seems that the charging limit is based on the software battery percentage rather than the hardware battery percentage.

Any insights or solutions would be appreciated. Thanks!

MatthiasKerbl, 2024-01-24:
Hi @jerrysearch,

Could please share a debug file the next time this happens so that we can assess this issue better? You can find a guide on [How to generate and share a debug file](https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/) on our blog.

Best,
Matthias

jerrysearch, 2024-01-29:
> Hi @jerrysearch,
> 
> Could please share a debug file the next time this happens so that we can assess this issue better? You can find a guide on [How to generate and share a debug file](https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/) on our blog.
> 
> Best, Matthias

Hi @MatthiasKerbl ,

This is a debug file. In order to upload attachments, the suffix name was changed to txt.

[aldentepro-debug-2024-01-29-11:32:19.pasta.txt](https://github.com/AppHouseKitchen/AlDente-Charge-Limiter/files/14079363/aldentepro-debug-2024-01-29-11.32.19.pasta.txt)

MatthiasKerbl, 2024-01-29:
Hi @jerrysearch,

looking at your debug file, everything seems fine. Could you test again, if AlDente does not use the Hardware Battery Percentage during normal use? Also, please keep in mind that during the Calibration Mode, AlDente uses the macOS battery percentage due to the limitations of some devices.

Best,
Matthias

MatthiasKerbl, 2024-06-18:
Hi @jerrysearch,

Any updates regarding your issue?

Best,
Matthias

jerrysearch, 2024-06-19:
> Hi @jerrysearch,
> 
> Any updates regarding your issue?
> 
> Best, Matthias

After updating to version 1.26.1, the problem has not appeared so far. The overall operation is in line with expectations. Thank you.

closed 2024-06-19 by MatthiasKerbl