AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1145: Crashes after turning off discharge in clamshell mode

deng-cy, 2024-01-07:
**Describe the bug**
When turning off discharge at clamshell mode, the software crashes. 

**To Reproduce**
Steps to reproduce the behavior:
1. Charge the battery to 100% 
2. Set limit to 70%
3. Use clamshell mode
4. Click "Discharge" to enable it
5. Click "Discharge" again to disable it

Note: I’m not sure if 1 and 2 are necessary. I confirmed 3, i.e., the software didn’t crash if the internal screen is on. 

**Expected behavior**
The software crashes

**Screenshots**
If applicable

https://github.com/AppHouseKitchen/AlDente-Charge-Limiter/assets/40417707/e7c32eaf-02f0-4b8f-a2a3-77b8de8b37f7

, add screenshots or screen recordings to help explain your problem.

**MacBook (please complete the following information):**
 - MacBook model A2992
 - macOS version 14.2.1
 - AlDente Pro
 - AlDente Version 1.24.1

MatthiasKerbl, 2024-01-29:
Hi @deng-cy,

Thank you for the bug report. We were able to fix this bug and will release an update soon.

Best,
Matthias

MatthiasKerbl, 2024-02-09:
Hi @deng-cy ,

We have just released AlDente 1.25 which includes a fix for this issue. To get AlDente 1.25 you just need to update within the app under Settings/General/Check for Updates.

Please report back if it works for you.

Best,
Matthias

closed 2024-02-09 by MatthiasKerbl