AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1146: Battery does not charge after AlDente uninstall

simonpra88, 2024-01-09:
Hi,
I uninstalled AlDente and now my Macbook Air M1 does not show the correct charging status in the menu bar anymore. In system preferences it correctly shows that the battery is charging when plugged in, the icon in the menu bar however shows the fully charged icon. 

[image]

Please help. I'm worried AlDente messed up my Mac's battery management. Will it damage my battery?

I followed all the steps here for a correct uninstall: https://apphousekitchen.com/how-to-uninstall-aldente-pro/ 

Thanks,
Simon

MatthiasKerbl, 2024-01-23:
Hi @simonpra88,

I am sorry to hear you are experiencing this issue, which should not happen especially when following our uninstallation guide.
Problems like yours are usually fixed by performing an SMC reset. This is done on Apple Silicon MacBooks by simply turning them off and closing the lid for about a minute. Afterward, your MacBooks battery should charge normally again.

Best,
Matthias

simonpra88, 2024-01-25:
Thanks Matthias, but tried that already several times without success. Problem still persists.

Any other suggestions to fix the problem besides reinstalling the whole OS? 
Do I need to worry, that my battery is overcharging and getting damaged because of this?

MatthiasKerbl, 2024-01-30:
Hi @simonpra88,

This is odd since all traces of AlDente are gone when you follow our uninstallation guide and perform an SMC reset.
A couple of more things you could try are the following:
- Try different charging cables and power adapters.
- Try different charging ports.
- Install AlDente again and set the charge limit to 100%.

> Do I need to worry, that my battery is overcharging and getting damaged because of this?

No you don't, Apple has safety measures in place to avoid this.

Best,
Matthias

MatthiasKerbl, 2024-02-20:
Hi @simonpra88,

Any update on this?
Also, we have just released AlDente 1.25.2 which has an updated debug file. If possible, please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on [How to generate and share a debug file](https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/) on our blog.

Best,
Matthias

MatthiasKerbl, 2024-04-12:
Hi everyone,

We have just released AlDente 1.26.1 which includes a fix for this issue.
To get AlDente 1.26.1 you just need to update within the app under Settings/General/Check for Updates.

Best,
Matthias

closed 2024-04-12 by MatthiasKerbl