AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1152: I didn't receive email with license

gornostay25, 2024-01-23:
@MatthiasKerbl
I bought license 20.01.2024 at 6:49 PM
I think I get typo mistake in email, so I register this "wrong" email and tried to resend license code using paddle.net help, but it doesn't work
Transaction no. #1204739286 PADDLE.NET* APPHOUSEKI

MatthiasKerbl, 2024-01-23:
Hi @gornostay25,

I am sorry you are having trouble with the buying process.
Please take a look at the following steps from our FAQ:

**I Did Not Get The License Code Or Any Receipt Emails After Buying AlDente Pro!**

There are multiple reasons why you did not get your license code after the purchase:

1. Sometimes it takes up to 30 Minutes for the purchase emails to go through.
2. The purchase emails end up in the junk email folder.
3. You had a typo in the email address during your purchase.
4. Failures of the payment networks/providers.

Please contact the support team of our payment provider Paddle at [assist@paddle.com](mailto:assist@paddle.com) and include the email address you used to purchase AlDente Pro. The Paddle support will try to solve your issue as fast as possible.

Cheers,
Matthias

gornostay25, 2024-01-23:
Thanks, I wrote an email to them yesterday, but they didn't answer to me

MatthiasKerbl, 2024-01-23:
> Thanks, I wrote an email to them yesterday, but they didn't answer

Perfect, they usually answer within a couple of days.

Best,
Matthias

closed 2024-01-23 by MatthiasKerbl