AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1154: Did not charge during sleep after wake forced 100%

axeII, 2024-01-23:
**Describe the bug**
As the title says, I was working on my MacBook Pro M3 in bed. I think the battery was around 35% before I went to sleep a put macbook to magsafe charging (hopefully debug logs show more). In the morning, I woke up and opened the laptop with magsafe charger still in. I noticed that the charging LED MagSafe color was orange and that the MacBook was charging. I expected that in the morning, the Mac would be charged to the limit of 75%. What surprised me even more was that the limit was set to 100%. It was difficult to change the limit; I had to stop charging by right-clicking, and then it was possible.

I checked, and calibration should not be running. I have one planned for next weekend...

**To Reproduce**
Steps to reproduce the behavior:
1. Use the mac to battery percentage 34%~
2. Close the lid of mac and plug in charger
3. After x hours wake the macbook (open lid) 
4. After wake mac starts charging and is set to 100%

**Expected behavior**
I expected that aldente will be charing to 75% instead of 100%. I did not set any calibration.

**Screenshots**
[image]
[image]
[image]
[image]

**MacBook (please complete the following information):**
 - Macbook pro m3 PRO
 - Sonoma 14.2.1
 - AlDente Pro 
 - AlDente 1.24.1
 

**Additional context**
[aldentepro-debug-2024-01-23-09:55:57.pasta.zip](https://github.com/AppHouseKitchen/AlDente-Charge-Limiter/files/14021503/aldentepro-debug-2024-01-23-09.55.57.pasta.zip)

MatthiasKerbl, 2024-01-23:
Hi @axeII,

Unfortunately, third-party apps, such as AlDente, are not able to run while the MacBook is asleep. Therefore, we had to come up with features to circumvent this limitation. One of these features is "Disable Sleep until Charge Limit". You can read more about this feature on our blog ["Feature Explanation: Disable Sleep until Charge Limit"](https://apphousekitchen.com/feature-explanation-disable-sleep-until-charge-limit/).

Best,
Matthias

axeII, 2024-02-22:
Got it! Thanks for letting me know. Closing this now. 

Btw 
> Unfortunately, the display will stay on for Apple Silicon based MacBooks. 

I wish this will be possible someday but I am guessing this needs to done by Apple.