AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1155: Charging is intermittent

ppmg77, 2024-01-24:
**Describe the bug**
I've set the limit to 70%. When I'm away and using my MacBook Pro M1, draining the battery below the limit.
Afterwards, I plug it in again and charging begins, but... Charging stops, charging begins again, charging stops, charging begins again, and so on until it reaches the set limit of 70%.

**To Reproduce**
Steps to reproduce the behavior (these are the steps I do and the behavior I'm seeing):
1. Charging till set limit of 70%
2. Using the MacBook Pro M1 on battery power
3. After using on battery, plug-in magsafe charger (this happens mainly during mornings when I'm working thus using the MacBook) and charging begins.
4. Charging will be intermittent. Every so many seconds charging disconnects and then connects again. This goes on till battery charge reaches the set limit.

**Expected behavior**
Draining the battery below the set limit in AlDente, plugging-in the magsafe charger and charging begins and continues in one go reaching the set limit and then stops charging.

**Screenshots**
None available (yet).

**MacBook (please complete the following information):**
 - MacBook model M1 Max
 - macOS version 14.2.1
 - AlDente Pro or AlDente Free? AlDente Pro (SetApp)
 - AlDente Version? 1.24.1
 - List of all enabled/disabled settings in AlDente:
    - Charge-tab: Only disabled "Stop Charging when app closed" and "Lock Screen when closed"
    - Features-tab: Sailing: 10% Enabled  /  Heat prot: 36C Enabled
    - Appearance: Only disabled "Hide Menu bar Icon"  /  "show on startup"  /  "Always Off"  /  "Show Dock Icon"
    - Schedule-tab: Active schedule for calibration once a month.

**Additional context**
None

MatthiasKerbl, 2024-01-24:
Hi @ppmg77,

I am sorry to hear about this issue and thank you for the detailed bug report.
If possible, please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on [How to generate and share a debug file](https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/) on our blog.

Best,
Matthias

ppmg77, 2024-01-24:
Dear Matthias,

Attached Bug Report.

Patrice G.
***@***.***

________________________________________________________

- Indien u mij wilt toevoegen aan een mailinglist, gaarne mijn mail-adres als BCC toevoegen,
dit voorkomt een hoop SPAM in mijn mailbox! Mijn dank hiervoor.

- In case that you want to add me to a mailinglist, please add my e-mail-address as BCC,
this could prevent a lot of SPAM in my mailbox! Thanks.

- En caso que me quiera agregar a una lista de mailing, por favor incluyame en CCO,
esto previene un montón de SPAM en mi mailbox! Muchas gracias.
________________________________________________________

Please consider your environmental responsibility. Before printing this e-mail message, ask yourself whether you really need a hard copy.

MatthiasKerbl, 2024-01-24:
> Dear Matthias, Attached Bug Report. Patrice G. ***@***.***
> […](#)
> ________________________________________________________ - Indien u mij wilt toevoegen aan een mailinglist, gaarne mijn mail-adres [...]

Hi @ppmg77,

Unfortunately, I can't find the attached file. You might need to go to the Github webpage and attach the file there.

Best,
Matthias

ppmg77, 2024-01-24:
Uploaded the Bug report.

MatthiasKerbl, 2024-01-24:
> Uploaded the Bug report.

Where did you upload the bug report?
It is not attached to your previous message.

ppmg77, 2024-01-24:
Uploaded to initial message.

MatthiasKerbl, 2024-01-29:
Hi @ppmg77,

It seems like the reason your system switches between charging and charging paused is because of the Heat Protection feature.
Please disable this feature and let us know if you are still experiencing this issue.
Also, the next update will add a hysteresis for the Heat Protection feature, therefore, it won't switch between charging and charging paused that often.

Best,
Matthias

ppmg77, 2024-01-31:
Ok, disabling this feature did the trick.

closed 2024-02-01 by MatthiasKerbl