AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1158: Dismissing pop-up does not work when clicking outside of the icon box in the menu bar

miallo, 2024-01-28:
**Describe the bug**
When clicking on the icon in the menu bar I expect it to toggle the menu and so to close it. That also works reliably when clicking inside of the area that gets highlighted when holding down, but you can also trigger the menu with the cursor at the very top of the screen, but that will be very unreliable and most of the times just flicker for a split second and keep the menu open.

**To Reproduce**
Steps to reproduce the behavior:
1. Clicking _above_ the icon the menu-bar should toggle the menu

**Expected behavior**
Instead of toggling the menu, it (most of the times) will only open the menu and when trying to close it, it flickers instead. Clicking on the icon area itself works as expected.

**Screenshots**
https://github.com/AppHouseKitchen/AlDente-Charge-Limiter/assets/6929993/fbb4901e-9573-46e6-b226-01ef2db33527

**MacBook (please complete the following information):**
 - MacBook model Mac14,9
 - macOS 14.3 (23D56)
 - AlDente Pro
 - AlDente Version 1.24.1
 - List of all enabled/disabled settings in AlDente:
   - Sailing mode
   - Menu Bar Show percentage
   - MagSafe LED Always off
   - PowerFlow Show in popover

**Additional context**
I could not find a reason why sometimes it seems to work to toggle and then again without moving the mouse it does not...

Maybe related to #984 in that it could be because it uses a custom container (It is only a guess, but with the default components I would expect the behaviour to work out of the box)

MatthiasKerbl, 2024-01-29:
Hi @miallo,

Thank you for the detailed bug report and interesting observation which I haven't noticed before. We will look into it and try to improve it with one of the upcoming updates.

Best,
Matthias

MatthiasKerbl, 2024-09-03:
Hi @miallo,

We have just released AlDente 1.28.1 which includes a fix for this issue. To get AlDente 1.28.1 you just need to update within the app under Settings/General/Check for Updates.
Please report back if it works for you.

Best,
Matthias

miallo, 2024-09-10:
That is indeed looking good :) Thank you very much for your work!