AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1161: Display does not turn off when lid is closed when "Disable Sleep until Charge Limit" is enabled

alexanderhorner, 2024-02-09:
**Describe the bug**
The Display does not turn off once the lid is closed when the Macbook is charging and "Disable Sleep until Charge Limit" is enabled. 

**To Reproduce**
Steps to reproduce the behavior:
1. Charge MacBook (Charge Limit must not be reached)
2. Enable "Disable Sleep until Charge Limit"
3. Close lid slowly

**Expected behavior**
The internal display should turn off once the lid is closed.

**MacBook (please complete the following information):**
- MacBook Pro 14" 2021
- macOS Sonama 14.1.2 (23B92)
- AlDente Pro 1.24.1
- Active Features: Disable Sleep until Charge Limit, Sail Mode 15%

MatthiasKerbl, 2024-02-12:
Hi @alexanderhorner,

Thank you for reporting this issue. We will try to do something about it in one of the upcoming updates.

Best,
Matthias

nathang21, 2024-02-20:
+1 i'm having the same issue, see attached for debug file, happy to reproduce or provide more details if needed. 
[aldentepro-debug-2024-02-19-21:55:01.pasta.zip](https://github.com/AppHouseKitchen/AlDente-Charge-Limiter/files/14338823/aldentepro-debug-2024-02-19-21.55.01.pasta.zip)

What i've noticed, is my Macbook struggles to charge while in clamshell mode when my display is on, but when the display sleeps, or I switch to my other device via my KVM switch, then it will adequately charge. This is WAI, and was the status quo before I started using AlDente. Note: This is due to the KVM switch which can power 2 devices via PD at once, can only supply up to ~56W at once to a single laptop, which is plenty for sustained use, and will also charge the laptop when either A) the display is off or B) i'm using the 2nd device. However with AlDente, option A no longer works. 

However WITH AlDente, if I walk away, even though sleep is disabled via "Disable Sleep until Charge Limit", my display will never turn off, and thus the Mac won't charge sufficiently to ever reach the charge limit for it to actually sleep. I've been hovering around 29% all day today, despite leaving the house for a few hours and not using my device for 6+ hours.

dreit-p, 2024-03-26:
Still find my laptop overheated with working display in closed state...
I hope that the author is working on a fix for the bug, because at the moment this program is harming, not helping.

MatthiasKerbl, 2024-03-26:
> Still find my laptop overheated with working display in closed state... I hope that the author is working on a fix for the bug, because at the moment this program is harming, not helping.

Hi @dreit-p,

We understand your frustration. This issue will become a priority and we will try to do something about it in one of the upcoming updates.

Best,
Matthias

dreit-p, 2024-03-26:
> Hi @dreit-p,
>
> We understand your frustration. This issue will become a priority and we will try to do something about it in one of the upcoming updates.
>
> Best,
> Matthias

Thank you Matthias! I'll be looking forward to it :)

MatthiasKerbl, 2024-04-12:
Hi everyone,

I have just noticed that there is already an issue open about this bug. Therefore, I will close this issue and will give updates about it in the other issue #998.

Best,
Matthias

closed 2024-04-12 by MatthiasKerbl