AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1162: App crashing / doesn't start anymore

Wastl2IT, 2024-02-10:
After updating AlDente Pro from 1.24.1 to 1.25, it doesn't start anymore. Clicking the app-icon, I can see the Symbol in the status-bar for half a second, then it closes again. There is no error message.

I'm using AlDente Pro 1.25 on a MacBook Pro (Retina, 15", mid 2015) running macOS Monterey 12.7.3.

armandoangulov, 2024-02-10:
I'm having the same issue with the same model MacBook Pro 15", mid 2015.
After the update the app doesn't start anymore. I used CleanMyMac to uninstall it and re install again the app open and show the initial prompt of welcome but as soon as I install the helper app the app crash.

Where can I download the  AlDente Pro 1.24.1? To install it until this get fixed.

MacBook Pro 15" mid 2015 - macOS Sonoma 14.3 (with OpenCoreLegacy)

cojamalo, 2024-02-10:
I have the same behavior for 1.25- crashes on attempt to start it- for anyone else who wants it to work again- just reinstall 1.24.1 again via Github release link https://github.com/AppHouseKitchen/AlDente-Charge-Limiter/releases/download/1.24.1/AlDente.dmg

armandoangulov, 2024-02-10:
I was wondering if it was the same installer for the pro version. I just installed it and activate my license. It working great. 
Looking forward for the version 1.25 get fixed.

idea-list, 2024-02-11:
Can confirm the same issue on the same type of the machine: MBP Pro 15" Mid 2015, macOS Monterey 12.7.3, AlDente Pro 1.25.

ludo12345, 2024-02-12:
Can confirm too : MBP Pro 15" Mid 2015

MatthiasKerbl, 2024-02-12:
Hi everyone,

Thanks for letting us know.
We are currently investigating this issue and are trying to fix it.

Best,
Matthias

Iam7heequatorsniveller, 2024-02-13:
I'm experiencing the same crash issue on a MacBook Pro 2015 mid running macOS Monterey 12.7.3 after updating to 1.25. Any help would be greatly appreciated!

MatthiasKerbl, 2024-02-13:
Hi everyone,

We have just released AlDente 1.25.1 which includes a fix for this issue. To get AlDente 1.25.1 you just need to update within the app under Settings/General/Check for Updates.
Please report back if it works for you and thank you for your patience.

Best,
Matthias

closed 2024-02-13 by MatthiasKerbl

cojamalo, 2024-02-13:
1.25.1 Confirmed working for my Macbook Pro Mid 2015- thanks for the fix!

Iam7heequatorsniveller, 2024-02-14:
Can confiremed that 1.25.1 fix this problem for my MBP 2015 mid, thanks!

ludo12345, 2024-02-14:
problem fixed !! thanks!!