AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1163: Cannot update: Error

martin-braun, 2024-02-10:
**Describe the bug**
An error occurred...

**To Reproduce**
Wait for the update notice.

**Expected behavior**
Should update.

**Screenshots**
[image]

**MacBook (please complete the following information):**
 - MacBook model MacBook Pro 16-inch, 2019
 - macOS version 13.5.2 (22G91)
 - AlDente Pro
 - AlDente Version 1.24.1

<details>
<summary>List of all enabled/disabled settings in AlDente:</summary>
[image]
[image]
[image]
[image]
[image]
[image]
</details>

**Additional context**

Also noticed this:
[image]
Funny™️

MatthiasKerbl, 2024-02-11:
Hi @martin-braun,

We have already fixed this issue and are not able to reproduce it anymore.
Do you still experience this issue?

Best,
Matthias

martin-braun, 2024-02-11:
Good job, it's fixed, thanks.

closed 2024-02-11 by martin-braun