AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1164: Sleep disabled, reinstalling helper locks the screen

dwoak, 2024-02-11:
Sometimes I notice that Sleep is disabled from the Apple menu. Usually I just reinstall the helper app, and the ability to Sleep my Mac is restored. However now, when reinstalling the helper app, my screen gets locked. I can install the helper app, but not without having to unlock my screen a couple of times during the reinstall process.
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. For me, just reinstall helper app when I noticed Sleep is disabled from the Apple menu.
2. Click Reinstall, screen locks.

**Expected behavior**
No Screen lock

**Screenshots**

**MacBook (please complete the following information):**
 - MacBook model: Apple M1 Pro
 - macOS version: Sonoma Version 14.3.1 (23D60)
 - AlDente Pro or AlDente Free: Pro
 - AlDente Version: 1.25
 - List of all enabled/disabled settings in AlDente:
 [image]

**Additional context**
Add any other context about the problem here.

MatthiasKerbl, 2024-02-12:
Hi @dwoak,

First of all, AlDente disabling Sleep of your MacBook is a feature and not a bug.
If you do not want AlDente to influence the Sleep behavior of your MacBook, you need to disable "Disable Sleep until Charge Limit". However, this will result in a different charging behavior. You can read more about it on our blog [Feature Explanation: Disable Sleep until Charge Limit](https://apphousekitchen.com/feature-explanation-disable-sleep-until-charge-limit/).

Regarding the "reinstalling helper locks the screen" issue, I am currently not able to reproduce this issue.

- Are you using an external display?
- Is your MacBook in Clamshell mode?

Best,
Matthias

dwoak, 2024-02-12:
Ah I'd totally forgotten about the Disable Sleep until Charge feature. Ignore me on that one.
I think I've refined it a bit. When sleeping has been disabled, the screen gets locked when trying to reinstall the helper. I'm using an external display in clam shell mode.

MatthiasKerbl, 2024-02-13:
Hi @dwoak ,

We have just released AlDente 1.25.1 which includes a fix for this issue. To get AlDente 1.25.1 you just need to update within the app under Settings/General/Check for Updates.
Please report back if it works for you.

Best,
Matthias

closed 2024-02-13 by MatthiasKerbl

dwoak, 2024-02-13:
Ah great, thanks for your help.