AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1166: Macbook Pro M1 Pro will not charge past 80% with or without Al Dente installed

cathaleoin, 2024-02-14:
**Describe the bug**
Macbook Pro M1 Pro will not charge past 80% with or without Al Dente installed.  Setting the limit to 100% (or anything over 80%) results in the charge stopping at 80%.  Starting a battery calibration also does not charge to 100%, and the battery stops charging at 80%. A pause icon is instead shown in Al Dente.  Following the steps described to uninstall Al Dente has resulted in the Macbook Pro being unable to charge past 80% even without Al Dente. 

On install, I've followed the instruction to disable "Optimized Battery Charging".  I have also disabled features such as low-power mode to avoid any potential conflicts.  

On uninstall, following the [instructions](https://apphousekitchen.com/how-to-uninstall-aldente-pro/) exactly results in the same behavior when Al Dente is not present; cannot charge past 80%. 

I have confirmed there are no other battery management applications installed in the system. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Al Dente
2. Click on 'Top Up' or 'Start Calibration'
3. Battery charged to 80% and stops. 
4. Uninstall Al Dente following the [instructions](https://apphousekitchen.com/how-to-uninstall-aldente-pro/).
5. Start charging Macbook Pro from mains.
6. Battery does not charge past 80%. 

**Expected behavior**
Battery should charge past 80% to the value set in Al Dente, and to 100% when uninstalled. 

**Screenshots**
Screen recording attached. 

**MacBook (please complete the following information):**
 - MacBook model: Macbook Pro M1 Pro 14 Inch 2021
 - macOS version: 14.3.1 (23D60)
 - AlDente Pro or AlDente Free? Al Dente Pro
 - AlDente Version? 1.25.1
 - List of all enabled/disabled settings in AlDente: 
 - - Stop Charging When Sleeping: Disabled
 - - Stop Charging When App Closed: Disabled
 - - Automatic Discharge: Disabled
 - - Disable Sleep Until Charge Limit: Disabled
 - - Lock Screen When Closed: Disabled
 - - Hardware Battery Percentage: Disabled
 - - Sailing Mode: 10%
 - - Heat Protection: 33
 - - Charge Limit: 100%
 - - Other cosmetic settings are visible in the screen recording

**Additional context**
1. I have attempted this twice; one with personalized settings, and another with clean settings by deleting the files described in the uninstall instructions.  Same result either way. 
2. I have restarted multiple times with no resolution. 
3. The only workaround is to completely shutdown the laptop, charge on mains, and that reaches 100%. Whenever the laptop is turned on, charging past 80% fails.

MatthiasKerbl, 2024-02-14:
Hi @cathaleoin,

I am sorry to hear about your issue and thank you for the detailed bug report.

Just to be sure, did you perform an SMC reset already? You can do that by turning off your MacBook completely, closing the lid, unplugging the MacBook, and waiting for at least 30 seconds.
Also, could you generate and share a debug file when you try to charge above 80%? You can find a guide on [How to generate and share a debug file](https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/) on our blog.

Best,
Matthias

cathaleoin, 2024-02-14:
Hi @MatthiasKerbl ,

Thank you for the swift reply.  I can confirm that I did attempt a SMC reset.  To be sure, I tried again today and can confirm that there is a limit still in place.  Oddly, it's not increased to 81% instead of 80% (but same story as it stops there). 

I'll share three debug files in the archive.zip that I created since flagging the issue yesterday: 
1. aldentepro-debug-2024-02-13-23/51/59.pasta - first debug file using my personal settings. 
2. aldentepro-debug-2024-02-14-00/09/14.pasta - second debug file after deleting Al Dente files and setting up again. 
3. aldentepro-debug-2024-02-14-19/29/15.pasta - third debug file generated just now (after a SMC reset attempt earlier today)
4. 
[Archive.zip](https://github.com/AppHouseKitchen/AlDente-Charge-Limiter/files/14285248/Archive.zip)

MatthiasKerbl, 2024-02-15:
Hi @cathaleoin,

Thanks for the debug files, we have looked at them and could unfortunately not find the cause of your issue and if it's related to AlDente.
However, please send a message to our [official support](https://apphousekitchen.com/support/). David will try to figure out what is going on with your system over there.

Best,
Matthias

MatthiasKerbl, 2024-02-20:
Hi @cathaleoin,

I am not sure if you have already reached out to our official support.
If not, we have just released AlDente 1.25.2 which has an updated debug file. If possible, please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on [How to generate and share a debug file](https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/) on our blog.

Best,
Matthias

cathaleoin, 2024-02-23:
Sure thing @MatthiasKerbl , and apologies for my delayed response.  I've just updated and pulled the debug file - PFA. 
[aldentepro-debug-2024-02-23-14:47:35.pasta.zip](https://github.com/AppHouseKitchen/AlDente-Charge-Limiter/files/14386475/aldentepro-debug-2024-02-23-14.47.35.pasta.zip)

cathaleoin, 2024-02-29:
Hey @MatthiasKerbl, any thoughts on this?  I've tried a few different things all without success.  The only way I seem to be able to get a charge over 80% is to shut off the laptop completely and charge.  I'm really looking to try and get back the extra 20% to no avail.

MatthiasKerbl, 2024-03-01:
Hi @cathaleoin,

Thank you for the update, we are still investigating but it seems as if something else than AlDente is prohibiting charging even though AlDente is trying to let your MacBook charge.
If we know more, I will let you know.

Best,
Matthias

cathaleoin, 2024-03-07:
Hi @MatthiasKerbl, I seem to have found a workaround that solved the issue.  I downloaded a few other apps that provide similar functionality, dropped their charge limit to 80%, and then increased it.  One of them seems to have worked, as now it seems I can charge to 100% with or without Al Dente again.

MatthiasKerbl, 2024-03-18:
> Hi @MatthiasKerbl, I seem to have found a workaround that solved the issue. I downloaded a few other apps that provide similar functionality, dropped their charge limit to 80%, and then increased [...]

Hi @cathaleoin,

Thanks for letting us know about your findings.
We don't know yet what the cause of this issue is (maybe a glitch in macOS or other charge limiters), however, we are quite certain to be able to fix it if it happens in the future, and these improvements will be implemented in the upcoming AlDente update.

Best,
Matthias

xinming, 2024-04-01:
May I ask which applications have you used to resolve this issue? @cathaleoin 
I am facing the same issue, and I tried using https://github.com/actuallymentor/battery to reproduce your workaround without success, thus would like some suggestions in the meantime. Thanks!

MatthiasKerbl, 2024-04-12:
Hi everyone,

We have just released AlDente 1.26.1 which includes a fix for this issue.
To get AlDente 1.26.1 you just need to update within the app under Settings/General/Check for Updates.

Best,
Matthias

closed 2024-04-12 by MatthiasKerbl