AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1174: Popover Missing Stats

itsfatal, 2024-02-20:
**MacBook (please complete the following information):**
 - MacBook Model (https://support.apple.com/en-us/HT201608): MacBook Pro (14-inch, 2021)
 - macOS Version (https://support.apple.com/en-us/HT201260): 14.3.1
 - Charger Model (Apple, Third Party, Wattage,...):
 - Charging Cable (MagSafe 2, MagSafe 3, USB-C, Apple, Third Party,...):
 - AlDente Pro or AlDente Free: Pro
 - AlDente Version: 1.25.2
 
I'm not seeing these stats in the popover for some reason. 
[image]
[image]

MatthiasKerbl, 2024-02-20:
Hi @itsfatal,

You have to open each category by clicking on the downward facing error to enable them in the popover.

Best,
Matthias

itsfatal, 2024-02-20:
Ahhh it will reflect whichever sections i have expanded. Apologies. Thank you!

closed 2024-02-20 by itsfatal

vladcudoidem, 2024-05-31:
Hi @MatthiasKerbl,

not gonna lie, this is very unintuitive. A simple fix would be to replace the up/down arrows on the top-right corner of each category with toggles, like the one next to "Show in Popover". As far as I can tell, no UI behaviour would have to be changed, only a view.

1. You can only click on the arrows anyway (clicking anywhere else on those card views does nothing).
2. Toggles have exactly the same number of states as the arrows, namely 2. So the change would be: **_arrow down -> toggle off_** & **_arrow up -> toggle on_**.

Does that make sense?

Greetings
Vlad

MatthiasKerbl, 2024-06-03:
> Hi @MatthiasKerbl,
>
> not gonna lie, this is very unintuitive. A simple fix would be to replace the up/down arrows on the top-right corner of each category with toggles, like the one next to "Show in [...]

Hi @vladcudoidem,

Thank you for your feedback and your suggestions. We will consider it and try to improve it in one of the upcoming updates.

Best,
Matthias