AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1178: Battery limit not working on bootcamp with macbook intel

fraferla, 2024-02-24:
- MacBook Model : MBP 2019 16'
 - macOS Version: macos sonoma 14.3.1 
 - AlDente Version: AlDente Pro 1.25.2
 - External Monitor connected: yes
 - MacBook used in Clamshell Mode (with the lid closed): yes

Hi, 

after updating to the 1.25 version, the battery limit does not work anymore with intel mode when I access Windows with bootcamp. The same issue occurs for version 1.25.1 and 1.25.2. Version 1.24.1 works fine.

I use the pc in Clamshell Mode.

Thank you

MatthiasKerbl, 2024-02-26:
Hi @fraferla,

I am sorry to hear you are having issues. Unfortunately, I am not able to reproduce this issue on my Intel MacBook.
Are you still experiencing this issue?
Do you have more details about how to reproduce this issue?

Best,
Matthias

fraferla, 2024-02-26:
Hi Matthias,

Thanks for your reply.

Yes, I tried right now and the problem still persist. 

Basically, the 1.25.2 version works regularly on my mac when I use iOS, but not when I use Windows on the bootcamp partion.

For instance, I set the battery charge limit to 65%, but when I run Windows, the charge level begins to rise again, instead of being steady at 65%.

Downgrading to the 1.24.1 version seems to solve the problem. 

If you need more details I am happy to help,

Best

edward-bailey, 2024-02-28:
I can confirm i get the same behaviour, the recent update seems to have regressed the Intel Mode Feature when in Bootcamp. Please let me know if I can assist in providing more details.

MatthiasKerbl, 2024-02-29:
Hi everyone,

Unfortunately, we are still unable to reproduce this issue. We have tested it on multiple MacBooks now including the 2019 16" MacBook with AlDente 1.25.2 and it works without any issues for us. However, we will continue looking into this and give updates here.

Best,
Matthias

MatthiasKerbl, 2024-02-29:
> I can confirm i get the same behaviour, the recent update seems to have regressed the Intel Mode Feature when in Bootcamp. Please let me know if I can assist in providing more details.

Hi @edward-bailey,

It would be helpful if you could share the following information:
**MacBook (please complete the following information):**
 - MacBook Model (https://support.apple.com/en-us/HT201608):
 - macOS Version (https://support.apple.com/en-us/HT201260):
 - Charger Model (Apple, Third Party, Wattage,...):
 - Charging Cable (MagSafe 2, MagSafe 3, USB-C, Apple, Third Party,...):
 - AlDente Pro or AlDente Free:
 - AlDente Version:
 - List or add screenshots of all enabled/disabled settings in AlDente:
 - Please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on How to generate and share a debug file on our blog(https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/).
 - External Monitor connected?
 - MacBook used in Clamshell Mode (with the lid closed)?

Best,
Matthias

SidSidSid16, 2024-03-01:
I am experiencing similar issues.

* MacBook Model: MacBookPro14,3
* macOS Version: 13.6.4
* Charger: Apple 87W USB-C
* AlDente Version: Free 1.25.2
* AlDente Settings: Limit = 49%, Launch at login enabled, Intel mode enabled
* No external monitor and not in clamshell mode

Setting the charge limit to 49% and after confirming that the battery level is at this percentage and charging has stopped, shutting down the Mac and booting into Bootcamp Windows 10, the charging resumes and goes past the limit. Once in macOS, the charging stops, even though the charge limit has been exceeded, it maintains the higher charge level. The charge limit also breaks when the Mac is shut down, as it resumes charging and exceeds the limit just as it does in Windows.

Please do let me know if you'd like any additional information.

MatthiasKerbl, 2024-03-01:
Hi everyone,

We were finally able to reproduce this issue and will fix it with the next update. In the meantime, you can circumvent this issue by following these steps:

- Disable the Intel Mode
- Go to Settings/Charge and enable Stop Charging When Powered Off
- Enable the Intel Mode again
Now, the charge limit should persist even in Bootcamp.

Best,
Matthias

fraferla, 2024-03-01:
Hi @MatthiasKerbl,

thank you. 

I can confirm that the temporary solution works for me.

edward-bailey, 2024-03-04:
Hi @MatthiasKerbl 

Apologies i wasn't able to come back in a timely manner with the details requested but I can confirm as with @fraferla that your workaround works for me too.

NeedNot, 2024-03-07:
Hi I'm expirencing the same problem on my macbook pro (mid 2015). It also started prob like a month ago after an update but I didn't care enough to do anything about it. I'm not sure what you mean by Intel mode, I can't find it in the menu.

MatthiasKerbl, 2024-04-12:
Hi everyone,

We have just released AlDente 1.26.1 which includes a fix for this issue.
To get AlDente 1.26.1 you just need to update within the app under Settings/General/Check for Updates.

Best,
Matthias

closed 2024-04-12 by MatthiasKerbl