AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1179: License Activation Failure Despite Stable Internet Connection

benkosa, 2024-02-28:
**MacBook:**
 - MacBook Model: MacBook Pro 14-inch, 2021
 - macOS Version: Version 14.2.1 (23C71)
 - Charger Model:Third Party, 100w
 - Charging Cable:USB-C
 - AlDente Pro or AlDente Free:AlDente Free
 - AlDente Version: 1.25.2
 - Licence email: kelt.bestofshare@gmail.com
 - List or add screenshots of all enabled/disabled settings in AlDente:
[image]

**Describe the bug**
I am unable to activate my license despite having a stable internet connection.

**Expected Behavior**
The license should activate successfully with a stable internet connection.

**Screenshots and Screen Recordings**
[image]
[image]

**Debug file**
[aldentepro-debug-2024-02-28-17:04:40.pasta.txt](https://github.com/AppHouseKitchen/AlDente-Charge-Limiter/files/14437343/aldentepro-debug-2024-02-28-17.04.40.pasta.txt)

**Attempted Solutions**
Deactivate and attempted to reactivate the license, but it failed due to an error stating "unable to deactivate license."

MatthiasKerbl, 2024-03-01:
Hi @benkosa,

issues like that are usually caused by VPNs or Firewall apps such as Little Snitch or LuLu.
Do you have any VPNs or Firewalls running?
If yes, please allow AlDente to check the license.

Best,
Matthias

MatthiasKerbl, 2024-06-18:
Hi @benkosa,

Any updates on your issue?

Best,
Matthias

closed 2024-08-12 by MatthiasKerbl