AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1182: Show "Remaining Battery Time" in popover

ttrepper, 2024-03-02:
I would find it very useful to see the reamining battery time (always) and would like to request to show it maybe within the bar or below (maybe configurable)

[image]

MatthiasKerbl, 2024-03-04:
Hi @ttrepper,

Thank you for your input.
In general, we are trying to keep everything organized and not overload our Interface. Therefore, we will keep most data in the data panel for now.

Best,
Matthias

MatthiasKerbl, 2025-04-15:
Hi @ttrepper,

With the latest update, we have made the data widget completely customizable, so that you can only select the data points you are interested in.

Best,
Matthias

closed 2025-04-15 by MatthiasKerbl