AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1186: MacOS 14.4 on OCLP breaks functionality

minorthird, 2024-03-09:
**MacBook (please complete the following information):**
 - MacBook Model 11,4, spoofing 15,2
 - macOS Version 14.4, OCLP 1.4.2
 - Charger Model Apple 
 - Charging Cable MagSafe 2
 - AlDente Free
 - AlDente Version: 1.25.2
 - List or add screenshots of all enabled/disabled settings in AlDente: All Disabled
 - External Monitor connected? No
 - MacBook used in Clamshell Mode (with the lid closed)? No

**Describe the bug**
AlDente no longer limits charging with MacOS 14.4 on OCLP.

MatthiasKerbl, 2024-03-11:
Hi @minorthird,

Unfortunately, OCLP is not officially supported and AlDente might not work properly or at all in some MacBook/macOS configurations.

Best,
Matthias

minorthird, 2024-03-13:
Ok, this bug seems specific to MBP 11,4—MBP 11,2 runs AlDente fine.

MatthiasKerbl, 2024-03-15:
> Ok, this bug seems specific to MBP 11,4—MBP 11,2 runs AlDente fine.

Hi @minorthird,

The general problem seems to be that OCLP tricks AlDente into thinking that it is running on a different MacBook model than it does. Therefore, AlDente is using the wrong technical pathways to control the charging behavior. Unfortunately, there is no easy fix for this issue and we are currently focusing on different improvements and features.

Best,
Matthias

AshokEmrys, 2025-05-18:
@MatthiasKerbl I have same model MacbookPro 11,4 and AlDente works fine with OCLP without spoofing the model no. However on spoofing it to MacbbookPro 15,2 it breaks. I think AlDente is trying to use functionality something that is specific MacbookPro 15,2 and it fails. I think this issue can be solved easily by having a hidden option that lets user select the correct macbook pro model number that Aldento has to use.