AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1187: Click on "Top Up" has no effect while calibrating

tisba, 2024-03-10:
**MacBook (please complete the following information):**
 - MacBook Model (https://support.apple.com/en-us/HT201608): **MacBook Pro M1 Pro**
 - macOS Version (https://support.apple.com/en-us/HT201260): **14.4**
 - Charger Model (Apple, Third Party, Wattage,...): CalDigit Element Hub (https://www.caldigit.com/thunderbolt-4-element-hub/)
 - Charging Cable (MagSafe 2, MagSafe 3, USB-C, Apple, Third Party,...): **USB-C**
 - AlDente Pro or AlDente Free: **Pro**
 - AlDente Version: **1.25.2**
 - List or add screenshots of all enabled/disabled settings in AlDente:
 - Please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on How to generate and share a debug file on our blog(https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/).
 - External Monitor connected? **Yes**
 - MacBook used in Clamshell Mode (with the lid closed)? **Yes**

**Describe the bug**
Clicking on "Top Up" has no effect while calibration is performed.

**To Reproduce**
Steps to reproduce the behavior:
1. While in calibration mode
2. Click on 'Top up'
3. Nothing happens

**Expected Behavior**
I'd at least expect a message, that first calibration has to stop, or something. Maybe it should be offered to stop calibrating, if "top up" is clicked?

[image]

MatthiasKerbl, 2024-03-11:
Hi @tisba,

Thanks for letting us know, we will do something about it in one of the upcoming updates.

Best,
Matthias

MatthiasKerbl, 2024-04-12:
Hi @tisba,

We have just released AlDente 1.26.1 which includes a fix for this issue. Now, when you click on Top Up while in Calibration Mode, a message will pop up.

To get AlDente 1.26.1 you just need to update within the app under Settings/General/Check for Updates.

Best,
Matthias

closed 2024-04-12 by MatthiasKerbl