AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1196: Macbok random restart/shutdown after installing aldente

wilwillyy, 2024-03-26:
Hi,
i uninstalled AlDente and now my Macbook Pro 2019, because my MacBook often turns off by itself, sometimes it just restarts, sometimes it turns off. this happened after I installed AlDente.
The battery indicator was also stuck, but it was fixed after nvram and resetting SMC.
but the random restart problem has not been resolved.
even though I have plugged in the electricity.
can you help me?

MatthiasKerbl, 2024-03-28:
Hi @wilwillyy,

Please fill out the following questions:

**MacBook (please complete the following information):**
 - MacBook Model (https://support.apple.com/en-us/HT201608):
 - macOS Version (https://support.apple.com/en-us/HT201260):
 - Charger Model (Apple, Third Party, Wattage,...):
 - Charging Cable (MagSafe 2, MagSafe 3, USB-C, Apple, Third Party,...):
 - AlDente Pro or AlDente Free:
 - AlDente Version:
 - List or add screenshots of all enabled/disabled settings in AlDente:
 - Please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on How to generate and share a debug file on our blog(https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/).
 - External Monitor connected?
 - MacBook used in Clamshell Mode (with the lid closed)?

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected Behavior**
A clear and concise description of what you expected to happen.

**Screenshots and Screen Recordings**
If applicable, add screenshots or screen recordings of the issue to help explain your problem.

**Additional context**
Add any other context about the problem here.

MatthiasKerbl, 2024-06-18:
Hi,

It's been a while since we last heard from you, so I'm gonna go ahead and close this issue for now. If you need to come back to it later, just let me know. I'm here to help and can easily reopen it for you.

Best,
Matthias