AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1198: Hardware Battery Percentage Inaccuracy (BUG)

dankiedan, 2024-03-27:
**MacBook:**
 - MacBook Pro (13-inch, 2019, Two Thunderbolt 3 ports)
 - Sonoma 14.3
 - 3rd Party 100W Charger
 - 3rd Party Belkin Charger rated for 240W
 - AlDente Pro
 - AlDente Version: 1.25.2
 - Check automatically, Launch at login, Stop charge when sleeping/power off, Hardware battery percentage, Sailing at 5%, Heat Protection at 40, Show percentage, icon style is MacOS coloured, show on startup, popup animation, show percentage, sensor data and power flow both showing on popover. The limit is set to 80% battery charge.

 - No external monitor
 - Macbook used to live in clamshell for a while but it is "on the go" 50% of the time.

**Describe the bug**
Hardware battery percentage reports 0 with a charging symbol. When the hardware battery percentage is disabled, the charging is stuck at 65%. Sometimes, it will show a number that is not 0, a reasonable number that is within the range of the actual macbook battery, however most of the times, it shows 0. Below in the screenshot, the green charging symbol is Al Dente, and the black one is native MacOS.

**To Reproduce**
Steps to reproduce the behavior:
1. Enable Hardware Battery Percentage.

**Expected Behavior**
A working hardware battery percentage.

**Screenshots and Screen Recordings**
If applicable, add screenshots or screen recordings of the issue to help explain your problem.

[image]
[image]
[image]

MatthiasKerbl, 2024-03-27:
Hi @dankiedan,

Thanks for sharing this detailed bug report.
If possible, please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on [How to generate and share a debug file](https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/) on our blog.

Best,
Matthias

dankiedan, 2024-03-27:
Thanks for the reply. It seems that I am facing other issues regading native MacOS, and it probably is not a bug on your end. 

To add to this, I did find however, that the power flow UI was a little janky and appeared to not display correctly. I believe seeing a GitHub Issue (that has been closed now) that mentioned this bug, but it seems to still appear for me on my version of Al Dente.

The bug is where the computer icon for the power flow appears to be somewhere completely random in the pop over.

Just thought I'd mention this bug also.

Thanks again,
Dank

MatthiasKerbl, 2024-08-12:
Hi @dankiedan,

Do you still experience the Hardware Battery Percentage issue?

Best,
Matthias

dankiedan, 2024-08-12:
> Hi @dankiedan,
> 
> Do you still experience the Hardware Battery Percentage issue?
> 
> Best,
> Matthias

It appears to be a hardware issue on my end. We can close this issue if you would like.

Thanks
Daniel

MatthiasKerbl, 2024-08-13:
> > Hi @dankiedan,
> > Do you still experience the Hardware Battery Percentage issue?
> > Best,
> > Matthias
> 
> It appears to be a hardware issue on my end. We can close this issue if you would like.
> 
> Thanks Daniel

Hi @dankiedan,

Thank you for the update. Feel free to add any upcoming information. For now, I will close this issue.

Best,
Matthias

closed 2024-08-13 by MatthiasKerbl