AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1202: Charging past set limit

Lasvad, 2024-03-31:
**MacBook (please complete the following information):**
 - MacBook Model 15inch, 2019
 - macOS Version (https://support.apple.com/en-us/HT201260): 14.4
 - Charger Model (Apple, Third Party, Wattage,...): Apple
 - Charging Cable (MagSafe 2, MagSafe 3, USB-C, Apple, Third Party,...): Apple USB-C
 - AlDente Pro or AlDente Free: Free
 - AlDente Version: 1.25.2
 - List or add screenshots of all enabled/disabled settings in AlDente:
 - Please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on How to generate and share a debug file on our blog(https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/).
 - External Monitor connected? No
 - MacBook used in Clamshell Mode (with the lid closed)? No

**Describe the bug**

Charging past limit 
Went to 90% (screenshot is 85% as some time has past)
[image]

Limit was set to 80%
[image]

Manage battery longevity is enabled. maybe cause of this since if Mac is fake reporting percentages by 10%, indicating 100% when really 90% is charged, if AlDante doesn't use this fake value but the relies on the real value, AlDante might change to 80% of current total capacity while Mac reports this as 90%
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Set limit to 80
2. Charge
3. Charged to 90%

**Expected Behavior**
Should not charge more then 80%

amlboeton, 2024-05-26:
yup this happened to my Ventura 13.6.7 as well

MatthiasKerbl, 2024-06-18:
Hi @Lasvad,

We have just released AlDente 1.27 which has big changes under the hood for Intel-based MacBooks. This should also make the charge limit more accurate.

Best,
Matthias

closed 2024-06-18 by MatthiasKerbl