AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1215: Charge limit changes "on it's own".

elihollander, 2024-04-21:
**MacBook (please complete the following information):**
 - MacBook Model (https://support.apple.com/en-us/HT201608): M1 Pro Max 16"
 - macOS Version (https://support.apple.com/en-us/HT201260): 14.4.1
 - Charger Model (Apple, Third Party, Wattage,...): Apple
 - Charging Cable (MagSafe 2, MagSafe 3, USB-C, Apple, Third Party,...): Apple
 - AlDente Pro or AlDente Free: Pro
 - AlDente Version: 1.26.1
 - List or add screenshots of all enabled/disabled settings in AlDente: Charging—all enabled; Sleep—all disabled; Sailing Mode—enabled 10%; heat protection—enabled; low power mode—disabled; 
 - Please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on How to generate and share a debug file on our blog(https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/).
 - External Monitor connected? yes
 - MacBook used in Clamshell Mode (with the lid closed)? no

**Describe the bug**
A clear and concise description of what the bug is.
I set the charge limit 56% and twice I have notice that the charge limit "changed" on its own to 51%
**To Reproduce**
Steps to reproduce the behavior: I have not been able to reproduce this
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected Behavior**
A clear and concise description of what you expected to happen.
I don't expect Al Dente Pro to change such a setting
**Screenshots and Screen Recordings**
If applicable, add screenshots or screen recordings of the issue to help explain your problem.

**Additional context**
Add any other context about the problem here.

MatthiasKerbl, 2024-04-22:
Hi @elihollander,

Please generate and share a debug file the next time this happens so that we can assess this issue better.

Best,
Matthias

MatthiasKerbl, 2024-06-18:
Hi @elihollander,

Any update on this matter?

Best,
Matthias

elihollander, 2024-06-18:
Hello Matthias,

No… that problem hasn’t occurred again. Perhaps it was “something” I inadvertently did. 

Overall, I really like your software and I have been recommending it to others (who have gone ahead and purchased the software). 

Thank you,

Eli

_____________________________________________________________

Eli Hollander, Professor Emeritus        831.464.0616 (voice)
   Film and Digital Media Department     
   University of California, Santa Cruz                                             
   Santa Cruz, CA  95064
***@***.***
http://film.ucsc.edu/

￼

closed 2024-06-18 by MatthiasKerbl

elihollander, 2026-10-22:
I am continuing to really appreciate your app; thank you!

But, I’m having problems with AlDente starting automatically when I restart my computer.  AlDente is in the startup list in System Preferences, but when I restart my computer, AlDente doesn’t start automatically. The “Launch at Login” option in AlDente prefs is greyed out. 

I am on Mac OS 27.

Thanks,

Eli
-----------------------------------
Eli Hollander
eMail: ***@***.***
Phone: 831.464.0616
-----------------------------------

MatthiasKerbl, 2026-10-27:
Hi @elihollander,

Thank you, glad you like AlDente!

Could you please try this: turn off "Launch at Login" in AlDente's settings, quit and restart AlDente, and turn it on again. Please also make sure AlDente is in your Applications folder, because Launch at Login only works from there. If the toggle is still greyed out, a screenshot of the settings would help.

Best,
Matthias

elihollander, 2026-10-29:
Hi Matthias,

I deleted AlDente from the ”Login Items & Extensions” and restarted my MacBook Pro M1, but the “Launch at Login” switch in AlDente Settings is greyed out. I am running the 1.40 version that was updated just a few days ago.

I am puzzled. 

Eli

_____________________________________________________________

Eli Hollander, Professor Emeritus        
   Film and Digital Media Department     
   University of California, Santa Cruz                                             
   Santa Cruz, CA  95064
***@***.***
http://film.ucsc.edu/
1.831.464.0616

￼

MatthiasKerbl, 2026-10-30:
Hi @elihollander,

Thank you for trying that!

AlDente never greys out this switch, so I think it is simply turned off (grey means off). Since you removed AlDente from the Login Items, could you please click on the "Launch at Login" switch in AlDente's settings once to turn it on, and then restart your Mac?

Please also update to AlDente 1.40.1 first: open the Dashboard, click on the gear icon in the bottom-left corner, and click on Check for Updates.

If the switch does not react when you click it, a screenshot of the settings would help a lot.

Best,
Matthias

elihollander, 2026-10-30:
> 8. Hi Matthias,

I updated Aldente. I then tried quitting AlDente and deleting AlDente from the “Login items & extensions” in system settings, but then when I start AlDente, it shows up again in the System Settings in the Login Items, with the “Login at Startup” switch still greyed out. Then, when I restart my computer, AlDente doesn’t start up automatically, and when I launch it manually, the “Launch at Login” switch is still greyed out. Mmmmm….

￼

￼

_____________________________________________________________

Eli Hollander, Professor Emeritus        
   Film and Digital Media Department     
   University of California, Santa Cruz                                             
   Santa Cruz, CA  95064
***@***.***
http://film.ucsc.edu/
1.831.464.0616

￼

MatthiasKerbl, 2026-10-31:
Hi @elihollander,

Thank you for trying that and for updating!

The entry that comes back in "Login Items & Extensions" is most likely AlDente's helper under "Allow in the Background". AlDente needs it to control charging, so please leave that one turned on.

Launch at Login is a separate entry under "Open at Login". Could you please try this:
1. Click on the "Launch at Login" switch in AlDente's settings once. Does it turn blue, or does nothing happen?
2. If nothing happens, open System Settings > General > Login Items & Extensions, click on the "+" under "Open at Login", and add AlDente from your Applications folder.

Your screenshots didn't come through via email, so if it still doesn't work, could you please add them directly here on GitHub?

Best,
Matthias

elihollander, 2026-10-31:
Okay… Solved, Matthias.

In the “Background App Activity” there are two entries for AlDente. One was switched on, the other was off. I turned the one that was off, on. and now, in the AlDente setting. the “Launch at Login” switch turned blue.

I haven’t tested it, but I am sure it will work. If not, you’ll hear from me. Thanks for your tenacious help, and Happy Halloween in Vienna.

Best,
Eli

_____________________________________________________________

Eli Hollander, Professor Emeritus        
   Film and Digital Media Department     
   University of California, Santa Cruz                                             
   Santa Cruz, CA  95064
***@***.***
http://film.ucsc.edu/
1.831.464.0616

￼

MatthiasKerbl, 2026-11-02:
Hi @elihollander,

Great to hear, and thank you for letting us know what fixed it!

If AlDente doesn't start at login after your next restart, just let me know.

Best,
Matthias

elihollander, 2026-11-03:
Quick note to let you know I restarted the computer and AlDente started just as expected. 

Thanks again,
Eli

_____________________________________________________________

Eli Hollander, Professor Emeritus        
   Film and Digital Media Department     
   University of California, Santa Cruz                                             
   Santa Cruz, CA  95064
***@***.***
http://film.ucsc.edu/
1.831.464.0616

￼