AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1217: Mac is broken! Battery is not charging anymore, after uninstalling helper and application

fedulovivan, 2024-04-23:
According to the provided steps I have changed charge limit back to 100%, then uninstalled helper from the app UI and finally app itself from the Applications folder. No running procesess in memory, no traces. But there is no charging now!

MatthiasKerbl, 2024-04-23:
Hi @fedulovivan,

I am sorry to hear that you are having issues with AlDente.
Please answer the following questions so that we can assess this issue better:

**MacBook (please complete the following information):**
 - MacBook Model (https://support.apple.com/en-us/HT201608):
 - macOS Version (https://support.apple.com/en-us/HT201260):
 - Charger Model (Apple, Third Party, Wattage,...):
 - Charging Cable (MagSafe 2, MagSafe 3, USB-C, Apple, Third Party,...):
 - AlDente Pro or AlDente Free:
 - AlDente Version:
 - List or add screenshots of all enabled/disabled settings in AlDente:
 - Please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on How to generate and share a debug file on our blog(https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/).
 - External Monitor connected?
 - MacBook used in Clamshell Mode (with the lid closed)?

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected Behavior**
A clear and concise description of what you expected to happen.

**Screenshots and Screen Recordings**
If applicable, add screenshots or screen recordings of the issue to help explain your problem.

**Additional context**
Add any other context about the problem here.

fedulovivan, 2024-04-23:
Sorry to be your free QA team.. Only weird "restart after all" action from Windows world helped to fix the issue (macbook had uptime 168 days before I did that). I think you should state about all low-level under-the-hood-magic at the very top of README and clearly list all risk user takes istalling your software.

MatthiasKerbl, 2024-04-24:
Hi @fedulovivan,

Without more information, it is difficult to say what went wrong with your system during the uninstalling of AlDente. I guess that something happened during the uninstallation which resulted in the system needing an SMC reset which is automatically done when restarting the MacBook.
Anyways, I am glad it is fixed now and thank you for your feedback.

Best,
Matthias