AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1218: "Stop charging when app closed" is disabled but computer won't charge when AlDente is not running

dingwen07, 2024-05-03:
**MacBook (please complete the following information):**
 - MacBook Model (https://support.apple.com/en-us/HT201608): MacBook Pro 13-inch (M1, 2020)
 - macOS Version (https://support.apple.com/en-us/HT201260): 14.4.1
 - Charger Model (Apple, Third Party, Wattage,...): Apple
 - Charging Cable (MagSafe 2, MagSafe 3, USB-C, Apple, Third Party,...): USB-C
 - AlDente Pro or AlDente Free: Pro
 - AlDente Version: 1.26.1
 - List or add screenshots of all enabled/disabled settings in AlDente: Stop charging when app closed is disabled
 - Please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on How to generate and share a debug file on our blog(https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/). [aldentepro-debug-2024-05-03-01:09:49.pasta.zip](https://github.com/AppHouseKitchen/AlDente-Charge-Limiter/files/15196403/aldentepro-debug-2024-05-03-01.09.49.pasta.zip)

 - External Monitor connected? No
 - MacBook used in Clamshell Mode (with the lid closed)? No

**Describe the bug**
Computer won't charge at all if AlDente is not running

**To Reproduce**
1. Install AlDente, make sure Stop charging when app closed option is disabled
2. Computer can charge when not reaching limits
3. Quit AlDente
4. Charging stopped
5. Uninstall Priviliged Helper Tool and reboot
6. Charging won't stop

**Expected Behavior**
Computer will charge when AlDente is off

**Additional context**
I tried reset all AlDente settings and removing Priviliged Helper Tool and removing `~/Library/Application Support/AlDente`. After this computer do charges, but after re-installing Priviliged Helper Tool, the issue persist.

acdb91, 2024-05-07:
I have the same issue:

- MacBook Pro 14 (2021)
- MacOS 14.4.1
- AlDente 1.26.1

MatthiasKerbl, 2024-05-08:
Hi @dingwen07 and @acdb91,

Thanks for letting us know about this bug. It will be fixed with the next update soon.

Best,
Matthias

dingwen07, 2024-06-10:
This problem also happens on a MacBook Pro Late 2021. AlDente will cause the computer unable to charge even after quit. Only way to solve is to discharge the battery to 20%, but the issue will come back after keep using AlDente.

MatthiasKerbl, 2024-06-18:
Hi everyone,

Thank you for your patience, we have finally released AlDente 1.27 which will fix this issue.

Best,
Matthias

closed 2024-06-18 by MatthiasKerbl