AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1232: Aldente broke my MB Pro Battery??

hansolowireless, 2024-05-25:
**MacBook (please complete the following information):**
 - MacBook Model (https://support.apple.com/en-us/HT201608):
 - M2 Pro
 - macOS Version (https://support.apple.com/en-us/HT201260):
 - Sonoma 14.5
 - Charger Model (Apple, Third Party, Wattage,...):
 - Charging Cable (MagSafe 2, MagSafe 3, USB-C, Apple, Third Party,...):
 - Usb-C
 - AlDente Pro or AlDente Free:
 - Free
 - AlDente Version:
 -
 - List or add screenshots of all enabled/disabled settings in AlDente:
 - Please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on How to generate and share a debug file on our blog(https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/).
 - External Monitor connected?
 - Charghing through external monitor after uninstalling Aldente
 - MacBook used in Clamshell Mode (with the lid closed)?

**Describe the bug**
Uninstalled Aldente Free and now my Macbook Pro does not charge, it shows battery percentage going down when connected to current.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected Behavior**
Macbook should charge

**Screenshots and Screen Recordings**
If applicable, add screenshots or screen recordings of the issue to help explain your problem.

**Additional context**
Add any other context about the problem here.

Kief5555, 2024-05-26:
You need to uninstall the helper if you havent already.

hansolowireless, 2024-05-26:
Nvm, may be due to another cause, but I thought Al Dente was the issue. 

You can close this ticket for now. Thanks.

closed 2024-05-26 by MatthiasKerbl