AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1234: The app no longer work after SO update

1111Al, 2024-05-28:
**MacBook Retina, 13-inch, Late 2013
 - MacBook Model (https://support.apple.com/en-us/HT201608):
 - macOS Version Big Sur 11.7.10
 - Charger Model 
 - Charging Cable MagSafe 2,
 - AlDente Pro or AlDente Free: Free
 - AlDente Version: 1.22

**Describe the bug**
A clear and concise description of what the bug is.

When the battery charge reaches the set level, it keeps charging as it there were no AlDente installed.

**Expected Behavior**
Battery charge stopping

After updating to Big Sur 11.7.10 the app stops working. I've unistalled the old version 1.22 along with assistant, and then installed the last one, but the problem persists.

MatthiasKerbl, 2024-06-18:
Hi @1111Al,

This might be due to a known issue that some users experience after a macOS update:
https://github.com/AppHouseKitchen/AlDente-Charge-Limiter/issues/768#issuecomment-1252390562

Best,
Matthias

MatthiasKerbl, 2024-08-12:
Hi @1111Al,

It's been a while since we last heard from you, so I'm gonna go ahead and close this issue for now. If you need to come back to it later, just let me know. I'm here to help and can easily reopen it for you.

Best,
Matthias

closed 2024-08-12 by MatthiasKerbl

1111Al, 2024-08-26:
> Hi @1111Al,
> 
> It's been a while since we last heard from you, so I'm gonna go ahead and close this issue for now. If you need to come back to it later, just let me know. I'm here to help and can easily reopen it for you.
> 
> Best, Matthias

Souds like it's all worked out after relaunched app and rebooted the Mac