AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1235: Low battery mode adjustment

htanx, 2024-06-05:
Turn off low power mode during the discharge phase of calibration mode

MatthiasKerbl, 2024-06-18:
Hi @htanx,

Interesting idea, we will look into it.

Best,
Matthias

pokono, 2024-08-09:
I have the same issue, also during automatic discharge.

I have my mac setup in high power mode when on power adapter and low power when on battery. When AlDente discharges the battery low power mode should be disabled temporarily until the process is completed, then restored to the original setting. This would give the automatic discharge better performance while retaining the low power mode for the usual battery usage.

I hope you guys can consider this for the next update.

cassus, 2026-10-10:
Yes, this is a great idea. I've been doing this manually for over a year now, and I'm happy to find that there's already a GitHub issue about it.