AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1237: [Feature request] High battery drain alert

SilverIce, 2024-06-14:
**Is your feature request related to a problem? Please describe.**
I'm getting frustrated whenever I launch power-hungry apps, but forgot to connect Mac to power supply. And intense battery drain degrades it.. 

**Describe the solution you'd like**
I would like to have an option to be alerted when something like this happening.

**Describe alternatives you've considered**
I use Parallels, so process monitoring isn't an option. That won't work

**Additional context**

MatthiasKerbl, 2024-11-21:
Hi @SilverIce ,

We have just released AlDente 1.29 with a new "Apps Using Significant Energy" panel within the Stats Section. To get AlDente 1.29 you just need to update within the app under Settings/General/Check for Updates.

Best,
Matthias

SilverIce, 2024-11-24:
@MatthiasKerbl thank you very much for this feature!

MatthiasKerbl, 2026-10-06:
Hi @SilverIce,

I am glad you like the "Apps Using Significant Energy" panel. With AlDente 1.39, we also redesigned it with Quit buttons, optional background processes, and improved app detection.

I will close this issue for now.

Best,
Matthias

closed 2026-10-06 by MatthiasKerbl