AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1244: Crash when MacBook is in sleep mode

AnotiaWang, 2024-06-19:
**MacBook (please complete the following information):**
 - MacBook Model (https://support.apple.com/en-us/HT201608): MacBook Pro 14' 2023
 - macOS Version (https://support.apple.com/en-us/HT201260): macOS 15 Developer Beta 1
 - Charger Model (Apple, Third Party, Wattage,...): MagSafe
 - Charging Cable (MagSafe 2, MagSafe 3, USB-C, Apple, Third Party,...): MagSafe 3
 - AlDente Pro or AlDente Free: AlDente Pro
 - AlDente Version: 1.27
 - List or add screenshots of all enabled/disabled settings in AlDente:

[image]
[image]
[image]

 - Please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on How to generate and share a debug file on our blog(https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/).

1. [aldentepro-debug-2024-06-19-21:09:58.zip](https://github.com/user-attachments/files/15901307/aldentepro-debug-2024-06-19-21.09.58.zip)
2. [macOS crash report](https://pastebin.com/2HYdZMXg)

 - External Monitor connected? 
Yes
 - MacBook used in Clamshell Mode (with the lid closed)?
Yes (It's in sleep mode, so I guess yes)

**Describe the bug**
AlDente crashes and exits in certain cases.

**To Reproduce**
Steps to reproduce the behavior: (I'm not sure if it's reproducible on other machines, but AlDente has crashed about 3 times on mine. I'll try to describe my use case)
1. Plug off the charger, external display and Internet cable
2. Connect MacBook to Wi-Fi
3. After some work, make the MacBook enter sleep mode by: pressing the fingerprint button, press Esc to close the screen, and close the lid.
4. After some time, plug in the charger, Internet cable and external display
5. Open the lid, and unlock macOS
6. macOS shows a dialog like "AlDente exited unexpectedly"

**Expected Behavior**
Should not crash

**Screenshots and Screen Recordings**

**Additional context**

Write, 2024-06-19:
I also have a lot of issue in clamshell mode + external screen connected to a dock since 1.27. I'll revert to 1.26 just to isolate the issue.

MatthiasKerbl, 2024-06-20:
Hi @AnotiaWang,

Thank you for your feedback, unfortunately, AlDente is not yet fully compatible with the macOS 15 Beta. Specifically one of the changes of AlDente 1.27 causes this issue. Therefore, we recommend not updating to macOS 15 until it is fully supported by AlDente or changing back to AlDente 1.26.1 which should not cause this issue.

Best,
Matthias

AnotiaWang, 2024-07-03:
I downgraded AlDente to 1.26.1, and chose "Skip this version" when AlDente asks me to update to 1.27, but still keep receiving new update popups. Is this a bug?

MatthiasKerbl, 2024-07-03:
> I downgraded AlDente to 1.26.1, and chose "Skip this version" when AlDente asks me to update to 1.27, but still keep receiving new update popups. Is this a bug?

Hi @AnotiaWang,

Do you have the "Check Automatically" setting enabled in AlDente Settings/General/Updates?

Best,
Matthias

AnotiaWang, 2024-07-03:
> Do you have the "Check Automatically" setting enabled in AlDente Settings/General/Updates?

Yes

MatthiasKerbl, 2024-07-04:
> > Do you have the "Check Automatically" setting enabled in AlDente Settings/General/Updates?
> 
> Yes

The popup should not appear anymore if you disable "Check Automatically".

Best,
Matthias

closed 2024-07-21 by AnotiaWang