AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1245: Updating to version 1.27 causes the screen to suddenly dim when it is closed and reopened, and AlDente will also stop running.

lucky9-cyou, 2024-06-19:
**MacBook (please complete the following information):**
 - MacBook Model (https://support.apple.com/en-us/HT201608): MacBook Air 15-inch, 2024 M3
 - macOS Version (https://support.apple.com/en-us/HT201260): Version 15.0 Beta (24A5264n)
 - Charger Model (Apple, Third Party, Wattage,...): Apple
 - Charging Cable (MagSafe 2, MagSafe 3, USB-C, Apple, Third Party,...): MagSafe 3
 - AlDente Pro or AlDente Free: AlDente Pro
 - AlDente Version: 1.27
 - List or add screenshots of all enabled/disabled settings in AlDente:
 [image]
[image]
[image]
[image]
 - Please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on How to generate and share a debug file on our blog(https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/).
https://drive.google.com/file/d/1EFPvNw5N85U7Vkjpu_qpBG881ZZjIb9K/view?usp=sharing
 - External Monitor connected? No
 - MacBook used in Clamshell Mode (with the lid closed)? No

**Describe the bug**
When I use version 1.27, closing and reopening my laptop causes Aldent to crash immediately, and the screen brightness dims to its lowest setting. This problem doesn't happen with version 1.26.1.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected Behavior**
A clear and concise description of what you expected to happen.
Fix it.

**Screenshots and Screen Recordings**
If applicable, add screenshots or screen recordings of the issue to help explain your problem.
https://drive.google.com/file/d/14zwLyFhg9Yl6wahsprZHGyvSqsgcyEnI/view?usp=sharing
**Additional context**
Add any other context about the problem here.

lucky9-cyou, 2024-06-19:
In the video, whenever I reopen my computer, the screen brightness automatically dims to its lowest setting. Additionally, all icons in the menu bar vanish because of an Aldente crash.

MatthiasKerbl, 2024-06-20:
Hi @lucky9-cyou,

Thank you for your feedback. This issue is caused by the macOS 15 Beta which is not yet supported by AlDente. Also, this seems to be a duplicate of issue #1244 where I also explained the issue in more detail. Therefore, I will close this issue for now and post updates on the other issue.

Best,
Matthias

closed 2024-06-20 by MatthiasKerbl