AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1246: Power flow thing is buggy

PeppinoTech, 2024-06-20:
**MacBook (please complete the following information):**
 - MacBook Model (https://support.apple.com/en-us/HT201608):MacBook Pro 13-inch,2020
 - macOS Version (https://support.apple.com/en-us/HT201260):Ventura 13.6.7 (22G720)
 - Charger Model (Apple, Third Party, Wattage,...):Third Party
 - Charging Cable (MagSafe 2, MagSafe 3, USB-C, Apple, Third Party,...):Third Party
 - AlDente Pro or AlDente Free:AlDente Pro
 - AlDente Version:1.26.1
 - List or add screenshots of all enabled/disabled settings in AlDente:
[image]
[image]
[image]
[image]

 - Please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on How to generate and share a debug file on our blog(https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/).
 - External Monitor connected? iPad thru the cable
 - MacBook used in Clamshell Mode (with the lid closed)? no

**Describe the bug**
A clear and concise description of what the bug is.
So.. The power flow thingy is buggy (you can see in the screen recording lower)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected Behavior**
A clear and concise description of what you expected to happen.

**Screenshots and Screen Recordings**
If applicable, add screenshots or screen recordings of the issue to help explain your problem.

https://github.com/AppHouseKitchen/AlDente-Charge-Limiter/assets/135860662/e44049bb-e322-487a-9d0f-2fe21ab3260d

**Additional context**
Add any other context about the problem here.

MatthiasKerbl, 2024-06-20:
Hi @PeppinoTech,

I am sorry to hear that the Power Flow feature is not working as expected for you.
As a first step, please update to AlDente 1.27 which has many under-the-hood improvements. Afterward, if it still does not work, we will look deeper into your issue.

Best,
Matthias

PeppinoTech, 2024-06-20:
ok i'll try

PeppinoTech, 2024-06-21:
[image]
:\

PeppinoTech, 2024-06-21:
@MatthiasKerbl

MatthiasKerbl, 2024-06-21:
Hi @PeppinoTech,

I am sorry to hear that the Power Flow feature is still not working as intended for you.
Could you please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on How to generate and share a debug file on our blog(https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/).

Best,
Matthias

MatthiasKerbl, 2024-08-12:
Hi @PeppinoTech,

It's been a while since we last heard from you, so I'm gonna go ahead and close this issue for now. If you need to come back to it later, just let me know. I'm here to help and can easily reopen it for you.

Best,
Matthias

closed 2024-08-12 by MatthiasKerbl

PeppinoTech, 2024-08-12:
Hi,@MatthiasKerbl. It's fixed btw I just really forgot about this issue lol