AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1250: Bootcamp charges always to full capacity since 1.27

quasiDigi, 2024-06-23:
**MacBook (please complete the following information):**
 - MacBook Model: MacBook Pro (15-inch, 2018):
 - macOS Version: 14.5 (23F79)
 - Charger Model: Third Party (Glyph TB3 dock)
 - Charging Cable: Third Party (TB3 cable)
 - AlDente Pro or AlDente Free: Free
 - AlDente Version: 1.27
 - List or add screenshots of all enabled/disabled settings in AlDente:
[image]
[image]
[image]
[image]
[image]
[image]

 - Please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on How to generate and share a debug file on our blog(https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/).
 - External Monitor connected? Yes
 - MacBook used in Clamshell Mode (with the lid closed)? No

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Set AlDente to 77%
2. Use Discharge option to empty battery to 77%
3. Restart computer in Windows (Bootcamp)
4. Use it for a while
5. Restart in macOs
6. Check battery status
7. See it's at 100% 

**Expected Behavior**
The battery charge was expected to stay at 77%

**Screenshots and Screen Recordings**
–

**Additional context**
–

8fdafs2, 2024-06-27:
> Unified the charging backend for all MacBooks with Intel CPUs. This means that "Stop charging when sleeping" and "Stop charging when powered off" are no longer required on these devices because the charge limit remains in these scenarios. This also means that "Intel Mode" is gone for good, since it is not needed anymore. For more info, please visit our [blog](https://apphousekitchen.com/adding-aldente-pro-support-to-16-new-intel-macbook-models-full-list-below/).

It appears that in version 1.27, the unified backend has a bug that causes Intel MacBooks to sometimes charge to 100%?

MatthiasKerbl, 2024-06-27:
Hi @quasiDigi and @8fdafs2,

Due to the unified charging backend and our attempt to streamline the user experience of AlDente throughout all MacBook models, the Intel Mode is gone now, and the automatic persistent charge limit is therefore only available with AlDente Pro. Suppose you want the charge limit to persist during Bootcamp. In that case, you can consider getting AlDente Pro for the automatic persistent charge limit or downgrading to AlDente 1.26.1 and continue manually using the Intel Mode.
You can get AlDente Pro on our [website](https://apphousekitchen.com/) and AlDente 1.26.1 on [GitHub](https://github.com/AppHouseKitchen/AlDente-Charge-Limiter/releases/tag/1.26.1).

Best,
Matthias

quasiDigi, 2024-06-28:
Thank you for your prompt answer. You might adapt your documentation, as this aspect is very unclear. Just adding your paragraph would solve it.

quasiDigi, 2024-06-28:
Solution proposed by @MatthiasKerbl , closing the ticket

closed 2024-06-28 by quasiDigi