AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1252: Installation completely broken by 1.27 update

christianh104, 2024-06-24:
**MacBook:**
 - MacBook Model: MacBookAir8,1
 - macOS Version: Sonoma 14.2.1
 - Charger Model: Apple
 - Charging Cable: Anker USB-C
 - AlDente Pro or AlDente Free: Pro
 - AlDente Version: 1.27
 - List or add screenshots of all enabled/disabled settings in AlDente:
[image]
[image]
[image]
[image]
[image]
[image]

 - Please share a debug file the next time this happens so that we can assess this issue better.
[aldentepro-debug-2024-06-23-23:58:21.pasta.txt](https://github.com/user-attachments/files/15951034/aldentepro-debug-2024-06-23-23.58.21.pasta.txt)

 - External Monitor connected? No
 - MacBook used in Clamshell Mode (with the lid closed)? No

**Describe the bug**
1. I cannot set the charge limit with the mouse, it just keeps switching itself to 15%.
2. I cannot enable sailing mode or heat protection mode, perform calibration, start discharge or top-up.
3. I can see the sailing mode, Intel mode, and heat protection mode icons in the popover despite them being "disabled".
4. Every time I open the popover I end up accidentally clicking away from it because it runs away from me when I move the mouse away from the menu bar. I believe this has been an issue for a while but I never really noticed because I set and forget it.
5. I cannot set the charge limit with the keyboard.
6. Despite the charge limit being stuck at 15% it is definitely charging when it is at 70+%.

**Screenshots and Screen Recordings**
https://github.com/AppHouseKitchen/AlDente-Charge-Limiter/assets/6159143/27c2ab72-0525-40a4-b2cd-cc23d0d68f3d
https://github.com/AppHouseKitchen/AlDente-Charge-Limiter/assets/6159143/f8736dd9-118d-49ab-acb9-c8a4f3ded40e

**Additional context**
Previously installed version was 1.26.1.
I did try restarting AlDente, but did not try to restart the Mac.
After resetting the settings in Al Dente and restarting it, everything was fixed except #4 above.
I also updated my 2023 M3 MacBook Pro to 1.27, and it worked perfectly except #4 above.

**Side note**
A friend and I both had our charge limits unexpectedly changed to 100% recently. For a while I thought Al Dente just screwed up and forgot the charge limit, so I set up a scheduled task to set the limit automatically every day in case it happened again and I didn't notice. I now think that this is due to an accidental right-click on the menu bar item. Start/stop charging is the default action but what is not spelled out there is that it changes the charge limit to do that, rather than just starting charging during sailing mode. This behavior combined with it being the default right-click action seems like a footgun for people who primarily use the software to avoid their battery being charged to 100%. Anybody who does this accidentally will definitely believe this is a failure on Al Dente's part, so it may not be the most sane default. Just a thought.

MatthiasKerbl, 2024-06-24:
Hi @christianh104,

Thank you for your detailed bug report and we are sorry for the inconvenience this caused.
You are the second user to report this issue where for some reason when installing AlDente 1.27 your installation slips through the safety measures and gets stuck. As you have already figured out, it can be fixed with a  simple settings reset.
Regarding point 4, this is a known issue and we will try to fix it with one of the upcoming updates.

Concerning your side note, thank you for the feedback, we will discuss it and consider changing the default right-click setting.

Best,
Matthias

christianh104, 2024-06-25:
Thank you for looking into this. I forgot that there is another small issue with the menu bar item: If I click the menu bar item with the mouse at the very top of the screen, it will not close the popover, it will just reopen it. This occurs on both of my MacBooks. I have also noticed this issue with another piece of software, DisplayLink Manager. Unfortunately they are not open source, otherwise it might be easy to track down something in common.

https://github.com/AppHouseKitchen/AlDente-Charge-Limiter/assets/6159143/91960ba0-be87-4ff5-a06c-6412b45be751

MatthiasKerbl, 2024-06-27:
Hi @christianh104,

Thanks for letting us know about this inconvenience. We will try to look into it.

Best,
Matthias

MatthiasKerbl, 2024-09-03:
> Thank you for looking into this. I forgot that there is another small issue with the menu bar item: If I click the menu bar item with the mouse at the very top of the screen, it will not close the popover, it will just reopen it. This occurs on both of my MacBooks. I have also noticed this issue with another piece of software, DisplayLink Manager. Unfortunately they are not open source, otherwise it might be easy to track down something in common.
> 
>  AlDente.3.mov

Hi @christianh104,

We have just released AlDente 1.28.1 which includes a fix for this issue. To get AlDente 1.28.1 you just need to update within the app under Settings/General/Check for Updates.
Please report back if it works for you.

Best,
Matthias

closed 2024-09-03 by MatthiasKerbl