AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1256: As a monitor tools, It seems write too much data on the disk

alexlabs, 2024-07-02:
Platform MacOS sonoma 14.5 
 
Please see the screenshot

alexlabs, 2024-07-02:
[image]

MatthiasKerbl, 2024-07-03:
Hi @alexlabs,

Thank you for your feedback. Depending on how long AlDente has been running on your system, it will have cumulated disk usage. In general, we try to keep AlDente as simple and resource-efficient as possible. Compared to some other menubar and monitoring apps, AlDente does not have more disk usage on my system. Nonetheless, we will try to improve it further in future updates.

Best,
Matthias

scenery, 2024-10-23:
@MatthiasKerbl  Same concern here. I noticed that my AIDente Pro (v1.28.5) has written over 10GB of data in less than 20 days. I'm not sure if this is related to logging. Perhaps it would be possible to provide users with an option to set the log level?

devklepacki, 2025-02-03:
For me as well, Aldente has the biggest disk usage from all my apps, at 6.2 GB write and 3.7 GB read. Why is that? The next regular app is Spotify with 1 GB wite / 7 GB read and apps like VPN of Safari.

Actually, it's not an issue in itself, as these are not astronomical values by any means. It's just a concern why is such app, that has nothing to do with actual data transfers, the highest disk usage of all apps?

foxhoundsk, 2025-03-01:
@alexlabs, does the write-much-data-on-disk problem still exist after using AlDente 1.31?

I think that the disk write is attributed to the sysfs write, which is how AlDente interects with Apple SMC (for charging controlling). Though a case that AlDente does do a lot of disk write would be the logging thing.

@MatthiasKerbl, how do you think?

scenery, 2025-03-01:
> [@alexlabs](https://github.com/alexlabs), does the write-much-data-on-disk problem still exist after using AlDente 1.31?

My situation: After upgrading to AIDente Pro 1.31.1, it has been running for 6 days and 12 hours, with 2.19GB written and 47MB read so far. While the write volume appears to have decreased somewhat, is there still significant room for further optimization?

closed 2025-03-10 by alexlabs

mailinglists35, 2026-03-23:
yup, no side effects. pro users would probably see paused stats; as a free user I don't see anything but charge cycles graph which should freeze. but yay no more writes AT ALL! and 12hr power is just above Spotlight at 5 percent and cpu at 1,7 percent (because it keeps retrying to access the sqlite db)

remember to quit AlDente when you're on battery so it doesn't drain unnecessarily. or find a way to automate start/quit it when plugged/when on battery.

devklepacki, 2026-03-25:
@MatthiasKerbl Sounds great! In which version should we expect this be updated? The last 1.37 or are we waiting for a new one now?

@mailinglists35 Thanks for notifying all the participants in the conversation about your (not necessarily important) opinion about other peoples' AI usage 😉 Probably you could let people live their lives however they want.

mailinglists35, 2026-03-25:
I am not an English native speaker but I still write in my own words. AI generated answers are simply offensive.