AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1258: Low and High-power mode switching

devMEremenko, 2024-07-03:
Hi,

Could you extend the list of options for the right-click on the menu bar?
I want to cycle between Low and High power modes, but currently, it cycles between Low and Automatic (even if HPM was active).

I would contribute, but it seems like the code is only available for the Pro version.

Thanks!

[image]

MatthiasKerbl, 2024-07-04:
Hi @devMEremenko,

Thank you for your feedback. We will consider it for one of the future updates.

Best,
Matthias

martin-braun, 2024-07-09:
What is HPM actually. I'm only aware of low power mode and turbo boost, but each of those can be controlled, separately. 

---

It it's not turbo boost, how can I enable HPM without any third-party app?

---

If it is turbo boost, we shouldn't cycle between those, should we? Low power mode sends instructions to other apps, allowing you to have fewer background tasks as well as the reduced screen brightness that comes with it. Turbo boost on the other hand allows a great peak performance of the CPU.

Real life example: I'm in the need of fast compile times for an Xcode project, but I'm out on the go. I enable Low Power mode, to reduce unnecessary work-load by background apps, but I still enable Turbo Boost to get good compile times. 

If this is a separate thing, it should be treated as such.

devMEremenko, 2024-09-05:
Hi @martin-braun,
You can learn more about HPM [here](https://support.apple.com/en-us/101613).

Note that it's only available on M-series Max models.
> You can use High-Power Mode on 14-inch MacBook Pro models with M3 Max and 16-inch MacBook Pro models with M1, M2 or M3 Max.

Apple describes HPM as a mode that can help with graphics-intensive tasks.
> High Power Mode can improve performance in graphics-intensive workflows

From my experience with the 16" M3 Max, it also helps with CPU-intensive workflows like building large Xcode projects faster. Large refers to first-time compilation time on M3 Max, approximately 80 seconds and above.

**How does it work?**
My assumption is that the system increases power limits in this mode and allows faster fan speeds.
While fans are going to 3k-4k RPM when Xcode is building, the real-time savings are measured in ### **seconds**.

I am fine with temporary noise. In fact, I even missed it because M-series chips are quiet. 🙂

martin-braun, 2024-09-08:
@devMEremenko Thanks for clarifying, I'm on Intel, no wonder I didn't hear about HPM.