AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1259: Misleading closed source

vractal, 2024-07-04:
If its not open source, you should not keep this stuff here with downloads from releases inside github WITHOUT ADDING A BIG DISCLAIMER on the readme stating that is not open source.  its not an empty repo either, you have code there, so its REALLY misleading.

Its ok to be closed source, but be HONEST about it. There is nothing in the readme saying its closed, and you have a big "Download from github releases" which imply there is source code available (you know, there is even a source code.zip there). Yes, you have a TINY statement in FAQ (not on readme) saying this is not open source anymore, but this is not enough and you know you are doing a misleading scammy thing for better marketing. And there is another comment in here from more than a year stating the same thing and you agreed with it but didnt change anything...

I was literally about to pay for PRO thinking this was open source until I realized its not and I had been deceived. I would probally have bought pro anyway if it was clear it was closed from the start, but will for sure not anymore.

martin-braun, 2024-07-09:
I'm a victim to this. I paid for this software to support the developer, but to be fair, I recalled it was closed source back when I bought that. However, to my "positive surprise" I learned it's "open-source" simply by looking at the repository at a later stage, just to feel dump for being that naive, thank you for bringing attention to this.

Like you, I have no problem for this being closed source, and I bought PRO while believing it to be closed-source that. No issue with that, I just feel tricked in my later false-positive discovery of this being "open-source".

MatthiasKerbl, 2026-10-08:
Hi @vractal and @martin-braun,

Thank you for your feedback, and sorry for the late reply. You are right, this wasn't clear enough. We have now updated the README: The section "About this repository" at the very top clearly states that AlDente is no longer open source, that the current version is proprietary and closed-source, and that the source code in this repository is legacy code from an older version.

I will close this issue for now.

Best,
Matthias

closed 2026-10-08 by MatthiasKerbl

martin-braun, 2026-10-09:
@MatthiasKerbl yes cool, I already moved to Linux and built my own little service that 'sails' my battery and also switches power profiles when plugged/unplugged. It doesn't have all the fluff Al'Dente has, but I had to leave macOS, so I made my own version and it gets the job done. Thanks for teaching me how to take care of my battery in theory. Long live Al'Dente and all our batteries.